UnchartedBull/OctoDash #1219: Choosing filament without unloading/loading

Its-my-right, 2020-11-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. If applicable also link other issues / bugs here.

**Describe the solution you'd like**
Hello,

Thank you very much for this great plugin, I really like it!
The possibility of adding custom actions is great ! (moreover, is it planned to be able to add a label in addition to the icons to navigate even better?)

On the other hand I wonder about the filament menu. From what I understand, it allows you to choose a target filament and then eject the current filament and insert the target filament.
This does not work well with the Prusa i3 MK3s in firmware 3.9.2, (I also tried by configuring octodash to make an M600), it sometimes activates the printer's loading / unloading functionality at the same time, or it detects crashes and puts the printer in "emergency" mode : /.
So, I got around this by creating 2 custom actions "filament load" (M701) and "filament unload" (M702). On the other hand, to choose the spool of filament currently loaded so that my printjob history is correctly saved, I still have to go through the filament menu, and hit "skip" on each screen that appears, being careful that the nozzle is cold, otherwise physical unloading / loading occurs (and therefore compatibility concerns with the Prusa).

My question is the following: would it be possible to be able to choose the spool of filament loaded in the printer, without launching the unload / load functionality?

Thank you :) !

**Link other projects**
If you want OctoDash to include other OctoPrint plugins or similar please link them here.

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2020-11-30:
If you just set your feedLength to zero it should skip over these (the feedLength is only used for filament change) plus set your purge distance to 0 as well.

Its-my-right, 2020-12-01:
Hi, thank you for your answer, it works when I click on green buttons on each screen which have one, and "skip" on the others. If I do "skip" on each screen, it doesn't work though.
I still enjoy using octodash with this workaround but I still think it would be great to have something like this :
- "Manual spool swap" : octodash trust you when you select the spool on touchscreen and returns to main screen
- "Assisted spool swap" : actual behaviour
That's just an idea :)

UnchartedBull, 2020-12-08:
I'm currently thinking of two possible ways:

- An additional button on the heat-up screen. Let's say you selected your filament and the printer is heating up (or waiting to heat up) - there will be an additional option that says "skip loading", which will directly put you back to the main menu (and cool down the nozzle)

- Hold down longer on one of the filaments. Let's say you touch the filament for ~2s and then you'll be redirected directly back to the main screen and filament manager will be updated with the latest filament.

Currently leaning towards the second option, what do you think?

Its-my-right, 2020-12-08:
Option 1 would add another button and I think one of the purpose of Octodash is to be quite simple/minimalistic right ?
So I think option 2 would be better. And maybe an highlight around the selected filament while touching it would be nice ? Because with the current version, I'm not always sure that I have selected the right filament (The following screen shows the color in a cirle, and the brand, but not full name, so with my two black Prusa filament (one petg and one asa) I am sure only on the last screen (which displays the entire filament name) that I've selected the right filament. And I'm thinking, if it is possible to "fast chose and update filament in filament manager" in a future release, this screen won't be displayed so highlighting the chosen filament may fix this

UnchartedBull, 2020-12-14:
Yeah I also had that happen today (two black filaments one PETG, one PLA+). Probably going to replace the circle with the icon from the list view. This won't take up any more space and would also tell the material which is really useful.

Highlighting the filament on "long" selection is something I'm having look at as well. I'm thinking something like a border that slowly extends. I'm thinking something similar to this: https://codepen.io/giana/full/xdXpJB. Once the border is fully done you'll be redirected back to the homescreen and the filament manager will be updated.

Its-my-right, 2020-12-15:
Indicate the filament type would indeed be very useful.
I looked at your border example and I think it would fit well with the "fast filament manager update without using Octodash internal unload/load" routine/behavior.
I would like to thank you very much for being careful and so positive towards suggestions for improvements, it's really nice especially since I imagine that you are developing Octodash in your spare time and that you have other activities alongside.
I enjoy using Octodash on a daily basis for the past few weeks, it's so nice and helpful :)

UnchartedBull, 2021-01-04:
Will be included in the next release. Feel free to build a "beta" version from master if you want to test things out :)

PR #1277 (2021-01-04): Rework filament change
Mostly behind the scenes code changes #1187 and some design tweaks. Should make #676 fairly easy now since filament manager is now behind a new interface.

Fixes #1193.
Closes #1219.
closed 2021-01-04 by UnchartedBull