UnchartedBull/OctoDash #1222: A clean install of OctoDash

Cosmo5125, 2020-12-02:
Hi there this is my first time installing OctoDash as im upgrading from TouchUi. 
The question that I have is that after doing a clean install of Octopi and doing an upgrade to ver 1.5.0 and following the instructions step by step using the auto install or the manual install, I cant change the log into desktop environment without installing the desktop version.
Is this correct? 
Also is there a way of booting straight into OctoDash without starting it manually through the desktop as I have to run the export DISPLAY command then octodash to start it or by using the touch screen which isnt always convenient.
Any advice would be great
Thank you

UnchartedBull, 2020-12-04:
Oh haven't noticed that update yet. I'll have a look at how to resolve this, you shouldn't need to install the Desktop as Ratpoison is able to take care of the window management.

UnchartedBull, 2020-12-04:
Ok I just updated to the latest raspi-config on my system as well. You need to go into `1 System Options` -> `S5 Boot / Auto Login` -> `B2 Console Autologin`. That shouldn't prompt you to install the Desktop (didn't for me). Please report back whether that works for you.

UnchartedBull, 2020-12-04:
I also updated the wiki just now: https://github.com/UnchartedBull/OctoDash/wiki/Installation#autologin

Cosmo5125, 2020-12-05:
Awesome thank you very much that works now

closed 2020-12-08 by UnchartedBull