UnchartedBull/OctoDash #1230: file sorting option to be saved automatically

Whitehawk2000, 2020-12-06:
The file sorting order always resets after leaving the filemanager. It would be great if octodash could remember the last option chosen by the user.

UnchartedBull, 2020-12-08:
You can define a default sorting in the settings for now.

It's a a good idea to remember the sorting for the whole session though. Will be included in one of the next releases.

Whitehawk2000, 2020-12-08:
> You can define a default sorting in the settings for now.

Just figured it out by accident yesterday evening. :-)

> It's a a good idea to remember the sorting for the whole session though. Will be included in one of the next releases.

I think so to, as it is more intuitive.

thegarbz, 2020-12-27:
@UnchartedBull Question: Why only save it for the session? It would be quite intuitive to save it permanently and just get rid of the options in the settings altogether.

UnchartedBull, 2020-12-30:
That probably is a good idea and should be fairly simple to do.

PR #2096 (2021-07-29): automatically save file saving order
Closes #1230
closed 2021-07-29 by UnchartedBull