UnchartedBull/OctoDash #1233: Independent homing of axes, shutdown button and ip-indicator.

Whitehawk2000, 2020-12-08:
As it makes more an more fun using octodash i think there are some suggestions to make it more comfortable.
1. Independent homing of X/Y-axis and Z-axis.
2. A Button to shutdown the pi from the touchscreen.
3. An indicator for the current ip-adress would sometimes be handy

UnchartedBull, 2020-12-14:
1 and 2 are already possible via Custom Actions: https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions

3 will be tracked in #1148.

closed 2020-12-14 by UnchartedBull

Whitehawk2000, 2020-12-15:
I know, that it is possible through the custom actions. But for No.1 that is not the logical place where the buttons have to be for an intuitive and comfortable user experience. Since octodash is leaned towards octoprint, the separate homingbuttons should be between the arrow keys, as it is in the control tab of Octoprint. 
Homing is an essential function and no custom action.

The same applies for No.2. The best logical place would be in the settings (maybe next to the "save" button)
Also No.3 should be found in the settings.