UnchartedBull/OctoDash #1234: Cannot find octopi.local

erra-ch, 2020-12-08:
**What doesn't work?**
Octodash says it can't connect to octopi.local

**What did you already try?**
I can access it with a browser, so the problem doesn't seem to be my router.

**General Information:**
 - Raspberry Pi 4
 - Octopi
 - OctoDash latest
 - OctoPrint Version 1.5.1

**Additional context**
The only difference is I set it up on my home network, does the API key change when switching network?
To be clear, OctoPrint is working as it should on my other network

UnchartedBull, 2020-12-14:
Could you maybe post a screenshot / photo of the error so I can check where exactly OctoDash has an issue?

erra-ch, 2020-12-17:
Will try to get a picture asap. But it is right after boot. 
After install it let me setup and everything, when I was done I think it said it couldn't find a printer which is obvious since my printer is in my garage (with another network). However when going back to the garage and booting I just get that one loading screen, "can't connect to octopi.local" or something similar. Meanwhile octopi is actually working _and_ accessible through octopi.local.

To be clear, I live in an apartment which is why I have a different network in my garage (~100m away)

UnchartedBull, 2020-12-19:
Does the pi have an internet connection if it's in the garage? Please aslo try pinging octopi.local from the Pi itself.

If OctoPrint and OctoDash are running on the same machine you can also just set your host in the settings to either localhost or 127.0.0.1 then OctoDash doesn't require the internet connection.

erra-ch, 2020-12-20:
Yes it has another network so I can remote control it. Haven't tried pinging octopi.local but as I said it works with web browser on both computer and phone.

Will probably reinstall octopi, octoprint and octodash. Hopefully it will work then. Would still be interested if you could figure out why I had problems though.

I'm reiterating now but the only difference is the network, I have both networks enabled in the octopi config which does work, that I am certain of. My gut feeling was probably wrong, even if the API key would change when switching network, wouldn't octodash just ask for the new one?

UnchartedBull, 2020-12-23:
If the API Key was wrong OctoDash would throw a number of 403 errors. To get behind this I'd need the exact error message, this way I can check which part of the code isn't happy with your setup. `Can't connect to ...` can only come up during initial setup. I guess best shot here is a screenshot, unless you got it working of course.

thegarbz, 2020-12-27:
From a networking perspective you really shouldn't use "octopi.local". Instead use "localhost". The reason being that octopi.local resolves to your primary external network interface whereas localhost will resolve to the loopback interface. 
If your wifi drops out or your network interface otherwise has a problem while you're using octopi.local any existing connections will get interrupted. I'm not sure if this would actually cause octodash to throw an error but it's good practice not to to use an external interface when you want to communicate with yourself. 

Also being able to connect to octopi.local from the web browser proves that the mDNS server (avahi-daemon) works on the Pi. It does not prove that the pi itself is correctly able to send out mDNS requests. The only way to prove that is to open up a console on the pi and try to ping octopi.local.

But regardless you should always use localhost or the ipaddress 127.0.0.1 to connect to yourself.

erra-ch, 2021-01-13:
I've been busy but yesterday I finally went to my garage, here's the pictures.

[image]
This is what I meant.

[image]
Also managed to get this once or twice, not sure how.

While I still don't know why it didn't work before, I'm happy to say it's working now.
Here's my story from start to finish:
1. Had been using octoprint for several months, works beautifully.
2. Took my raspi home to setup octodash since I had purchased a touch screen , everything seemed to work, tough I obviously couldn't test it since I don't have my printer at home. So this was done at home, with network "**A**"
3. Took the raspi to my garage, plugged in my printer and started it, with network "**B**". So no changes except for the network has been made.
4. It did not work, I opened this issue. (Octodash didn't work, the printer and octoprint still worked perfectly)
5. I tried to ping octopi.local as you said, didn't work. It was not setup with localhost so I tried to change the config file, pretty  sure I didn't mess up but it didn't seem to work. (I rebooted)
6. At this point I figured I might as well just reinstall it, and perhaps set it up properly from the beginning. So I tried your uninstall script, which didn't work.
7. Since I am a beginner I looked up how to delete the octodash package. I did and then installed it again. Still didn't work, looked at the config file which was unchanged.
8. Finally I deleted the config file, reinstalled octodash and it finally let me set it up again. This time I chose localhost.
9. Now it seems to work, didn't try to print but I tried moving my Z-axis and it worked.

Hopefully I won't have more problems, Octodash is looking great and I'm excited to see it in action!

thegarbz, 2021-01-13:
@erra-ch that all points to an underlying issue with Octopi / Linux networking. "getaddressinfo EAI_AGAIN" is the error returned when the DNS server cannot resolve the requested name. This is corroborated with your inability to ping octopi.local

Unfortunately that's not an Octodash issue, there's an underlying problem with your Octopi that multicastDNS addresses (those ending with .local) are not resolving. Using "localhost" works around this issue since localhost is not multicastDNS but the problem likely still remains. There's a few things we can do to look at the underlying network. 

1. Connect to octopi via ssh. 
2. Run "ping octopi.local" to verify it's not working and "ping localhost" to verify it is.
3. Run "ifconfig" and check what the ip address is. This will be the "inet" address for eth0 if you use a wired network or wlan0 if you use a wireless network. Make sure the broadcast address has the same first numbers as the inet address. Importantly if you don't use either wired network or wireless you should **not** see an inet entry. If you do then that could be the source of the problem. 
```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.116  netmask 255.255.255.0  broadcast 192.168.2.255
```
4. Run "cat /etc/nsswitch.conf" and look for the line which says hosts: It should have mdns4_minimal in the list
```hosts:          files mdns4_minimal [NOTFOUND=return] dns```
5. Run "systemctl status avahi-daemon" this will give you the output of the mDNS daemon. It should say active (running) and you should see it join and register your address on the active interface. For me below it's the line which says "Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.2.116." and the "registering new address" line 2 lines further down. 
```
● avahi-daemon.service - Avahi mDNS/DNS-SD Stack
   Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-01-11 19:17:06 CET; 1 day 16h ago
 Main PID: 350 (avahi-daemon)
   Status: "avahi-daemon 0.7 starting up."
    Tasks: 2 (limit: 4915)
   Memory: 1.5M
   CGroup: /system.slice/avahi-daemon.service
           ├─350 avahi-daemon: running [octopi.local]
           └─391 avahi-daemon: chroot helper

Jan 11 19:17:12 octopi avahi-daemon[350]: New relevant interface wlan0.IPv6 for mDNS.
Jan 11 19:17:12 octopi avahi-daemon[350]: Registering new address record for fe80::f382:254a:ec:dd46 on wlan0.*.
Jan 11 19:17:15 octopi avahi-daemon[350]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::f382:254a:ec:dd46.
Jan 11 19:17:15 octopi avahi-daemon[350]: Joining mDNS multicast group on interface wlan0.IPv6 with address fd00:8494:8c16:dff2:e16:4f6a:7437:fc84.
Jan 11 19:17:15 octopi avahi-daemon[350]: Registering new address record for fd00:8494:8c16:dff2:e16:4f6a:7437:fc84 on wlan0.*.
Jan 11 19:17:15 octopi avahi-daemon[350]: Withdrawing address record for fe80::f382:254a:ec:dd46 on wlan0.
Jan 11 19:17:17 octopi avahi-daemon[350]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.2.116.
Jan 11 19:17:17 octopi avahi-daemon[350]: New relevant interface wlan0.IPv4 for mDNS.
Jan 11 19:17:17 octopi avahi-daemon[350]: Registering new address record for 192.168.2.116 on wlan0.IPv4.
Jan 13 11:31:20 octopi avahi-daemon[350]: Registering new address record for fd48:72f2:bede:1:5765:8b43:6cee:637 on wlan0.*.
```

UnchartedBull, 2021-01-23:
Thanks for chiming in here @thegarbz!

closed 2021-02-13 by stale[bot]