UnchartedBull/OctoDash #1242: Connection to OctoPrint timed out

pietchaki, 2020-12-13:
**What doesn't work?**
OctoDash boots up to the "Connecting" screen, then throw a message saying "Connection to OctoPrint timed out"

**What did you already try?**
I made sure Octoprint is up by logging into it via browser. Also tryed to reboot into safe mode but the same problem there.

**General Information:**
 - Hardware Raspberry Pi 3B+
 - OS Info OctoPi 0.17
 - OctoDash Version 2.1.2 (not sure about version, runned installation script as in the wiki yesterday, Deceber 12 2020)
 - OctoPrint Version 1.5.2

**Additional context**
I was running Octoprint with OctoScreen, and wanted to check OctoDash.
So I first made a Backup of Octoprint, and prepared a new SDCard with OctoPi.
Installed the new SDCard, and restored octoprint configurations (and plugins) into it.
Then installed the display drivers and finally run the OctoDash install script.
It booted and entered into the configuration menu. There I used the button to get the API KEY and confirmed from a browser logged into octoprint. All seemed OK and it was late into the night so I powered off the octoprint from browser.
today I powered on the printer, octoprint seems to be working no problem, but OctoDash can't connect to it.
Couldn't find any report of a similar problem, and I don't know what config or log files to look for. 

I really don't want to start all over again, display drivers for my unit are a real pain to install and make it work...

pietchaki, 2020-12-14:
Update:
Found the reason why it wasn't connecting... It was trying to connect to "http://octopi.local:80/api", but couldn't find it, I changed to "http://localhost:80/api" and it is working now.

Maybe the configuration proccess could try and use "localhost" instead of "octopi.local" for default, if octoprint is running on the same raspberry pi (I believe it is the most common) it will be better, If localhost didn't work, then try to use octopi.local

UnchartedBull, 2020-12-14:
Did you choose octopi.local from the list of available instances? If so there is nothing I can do here really. `octopi.local` is the advertised URL from OctoPrint which I'm using (via then bonjour service). This is the one that will be used if you click on the entry. Trying localhost automatically might be confusing for people with multiple instances. Glad you figured it out though!

closed 2020-12-14 by UnchartedBull

pietchaki, 2020-12-15:
Yes, it was from the list on the configuration steps.
What I do not understand is that for 1 session only it works without any problem, but after a reboot it doesn't. Maybe this first session is not using the reported URL? Maybe on second session it is getting something wrong for the URL?

As I didn't found other reports of this problem, it may be a problem of my current setup... I just don't know how that can be as I can reach octopi.local from the raspi and from other devices.

UnchartedBull, 2020-12-19:
Most likely some fishy network stuff. I can imagine the Pi not having internet access for the first 30 seconds (which does make OctoDash fail). If you use localhost the Pi won't use the internet at all and thus doesn't rely on the interface being up .During setup  it had some more time to gain a internet connection which made it work probably.