UnchartedBull/OctoDash #1243: Http failure resonse for...

TobsBobs, 2020-12-13:
Hello, I have installed octodash and have recently been getting the following error messages.

[image]
[image]
[image]

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:83.0) Gecko/20100101 Firefox/83.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3959967744
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
octodash: 2.1.2

What can I do to remove the error message?

Thanks for this great software and your help in this matter.

UnchartedBull, 2020-12-14:
Have you tinkered with the permission system in OctoPrint recently? 403 FORBIDDEN means that the API key for OctoDash isn't allowed to access the requested resource (printer status in this case). Either check the permissions for OctoDashs API Key or create a new one. Let me know if this worked.

TobsBobs, 2020-12-15:
Thanks, it was the API key.

closed 2020-12-15 by TobsBobs