UnchartedBull/OctoDash #1260: Error when updating to latest version

ZombiesLoveMe, 2020-12-22:
**Describe the bug**
when trying to update octodash via CLI it fails after 10 minutes, when updating via the touchscreen it freezes for 5 hours then crashes

**To Reproduce**
Steps to reproduce the behavior:
via CLI run command wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/update.sh | bash
via touch screen run the update 

**Expected behavior**
update to be preformed and annoying red dot go from the gear

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
Pi 4 with 5 inch Elecrow touch screen

**Additional context**
error from CLI when updating

`wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scr                                                                                                                ipts/update.sh | bash
Updating OctoDash
octodash.deb                                      0%[                                                                                                        ]  50.57K   175 B/s    in 4m 55s
[sudo] password for pi:
(Reading database ... 87611 files and directories currently installed.)
Preparing to unpack octodash.deb ...
Unpacking octodash (2.1.2) over (2.1.0) ...
dpkg-deb (subprocess): cannot copy archive member from 'octodash.deb' to decompressor pipe: unexpected end of file or stream
dpkg-deb (subprocess): decompressing archive member: lzma error: unexpected end of input
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive octodash.deb (--install):
 cannot copy extracted data for './opt/OctoDash/libGLESv2.so' to '/opt/OctoDash/libGLESv2.so.dpkg-new': unexpected end of file or stream
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Errors were encountered while processing:
 octodash.deb
Done. Restart your Raspberry Pi to start the newest version!
`

UnchartedBull, 2020-12-23:
Have you installed the [update script](https://github.com/UnchartedBull/OctoDash/wiki/Update#in-app-update) to enable the In-App Updates?

The error looks like a damaged package to me. Could you try running the update script again and check whether that fixes the issue? If that doesn't work please try removing `./opt/OctoDash/libGLESv2.so` and update again (please make a backup before as this could possibly break OctoDash). Another option would be to remove octodash and then install it again.

```
killall octodash
sudo dpkg -P octodash
```

ZombiesLoveMe, 2020-12-23:
> Have you installed the [update script](https://github.com/UnchartedBull/OctoDash/wiki/Update#in-app-update) to enable the In-App Updates?
>
> The error looks like a damaged package to me. Could you try [...]

thanks for this i tried fixing using the above but ended up reinstalling issue now is that its asking for a port and when closing the card it sits on connecting

UnchartedBull, 2020-12-23:
Port should be either 5000 or 80

ZombiesLoveMe, 2020-12-23:
> Port should be either 5000 or 80

[image]

Slight issue being I have no where to put a port 🤣🤣

UnchartedBull, 2020-12-23:
Please edit your config manually (https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#manually-adjusting-the-settings) and make sure that url looks similar to: `http://localhost:5000/api/`

ZombiesLoveMe, 2020-12-28:
> http://localhost:5000/api/

thanks for your help this is now fixed :D

closed 2020-12-30 by UnchartedBull