UnchartedBull/OctoDash #1261: OctoPrint do not show up and the screen showing white lines...

arut16, 2020-12-22:
**What doesn't work?**
On a functionnal OctoPrint setup, I'm trying to install octodash with a 5" capacitive screen [from here](https://fr.aliexpress.com/item/4000285091974.html?spm=a2g0s.9042311.0.0.570d6c37bRDc6d).
Now Octoprint do not showUp and the screen do not show octodash but some mystical lines for me...
[image]

**What did you already try?**
I tryed [this complete tutorial](https://induste.com/threads/tuto-installation-doctodash-sur-raspberry-pi-avec-ecran-local.635619/) when I'm tryed bash installation for automatic installation (bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)

**General Information:**
 - Hardware : RPi 4 Go with 800x460 screen [from here](https://fr.aliexpress.com/item/4000285091974.html?spm=a2g0s.9042311.0.0.570d6c37bRDc6d)
 - OS Info : Raspbian Buster with OctoPi 0.17.0
 - OctoDash Version : 2.0.0 when 2.1.7 (?) with auto script ?
 - OctoPrint Version : 1.5.2

**Additional context**
[image]

UnchartedBull, 2020-12-23:
Hmm I can't really see anything from the picture you posted since it doesn't contain any errors. Have you enabled automatic login for Pi like described [here](https://github.com/UnchartedBull/OctoDash/wiki/Installation#autologin)? Have you checked the [troubleshooting guide](https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting)?

arut16, 2020-12-23:
I ended up dropping this issue. And started again from the beginning with a fresh octroprint installation and everything installed correctly with bash auto script and works perfectly.

closed 2020-12-23 by arut16