UnchartedBull/OctoDash #1269: Possible to have OctoDash on a headless pi with VNC/RDP viewed from a tablet app?

daviiiiiid, 2021-01-01:
**What doesn't work?**
I've had success using this tablet as a screen for other pi projects but in this case, OctoDash never really shows up.

**What did you already try?**
Tried logging in via RDP and VNC.

**General Information:**
 - Hardware Huawei MediaPad T5
 - Raspberry Pi 3B+
 - OctoPrint 1.5.2 Python 2.7.16 OctoPi 0.17.0
 - OctoDash latest version as of today installed.

On VNC, I get "Cannot currently show the desktop"
On RDP, at first I saw that Ratpoison was starting at the top right corner, and then it stays on a black screen.

Is this not possible with this setup? I know its a bit complicated but I had this extra tablet and no spare monitor or pi touchscreen so thought I could make it a project.

UnchartedBull, 2021-01-02:
I don't exactly know how ratpoison does handle VNC, since you don't have an actual screen attached. Have you opted for the auto start during installation?

It should definitely be possible though!

daviiiiiid, 2021-01-02:
Yeah I did select autostart. Not sure what else I can try.. So far I gave up and just mounted the tablet and using the browser to octoprint.. really not as sleek but it does the job..

FrankX0, 2021-01-04:
I have a screen connected and run VNC: both show OctoDash correctly.
So your issue seems not to be cause by VNC.
My guess is your desktop environment is not running.

UnchartedBull, 2021-01-04:
Probably the best way here is to get this running on a screen (HDMI or anything really) and then put in the VNC part. Another option might be to install the fully fledged raspbian desktop distro and then run OctoDash at boot automatically there.

daviiiiiid, 2021-01-04:
> I have a screen connected and run VNC: both show OctoDash correctly.
> So your issue seems not to be caused by VNC.
> My guess is your desktop environment is not running.

Can't believe it was this simple. I had desktop GUI ENABLED but not INSTALLED.... I installed the desktop environment and now VNC Viewer works..

Now the problem is that on an Android tablet, it only uses touch to move the mouse and click and not actually work as a touch screen. That is highly inconvenient and looking for a solution!

FrankX0, 2021-01-05:
This behavior is due to the nature of VNC, as it emulates a mouse, not a touchscreen.
I don’t think you can change this behavior.

UnchartedBull, 2021-01-06:
Closing here since I can't really do anything about this unfortunately :/

closed 2021-01-06 by UnchartedBull

ipetrovits, 2021-11-29:
> This behavior is due to the nature of VNC, as it emulates a mouse, not a touchscreen. I don’t think you can change this behavior.

By the way it is supported on iOS version, but missing in Android client.