UnchartedBull/OctoDash #1272: OctoDash DisplayLayerProgress 404 error after install

MrEdskii, 2021-01-02:
OctoDash version 2.1.2
OctoPrint version : 1.5.2
OctoPi version    : 0.17.0
Raspberry Pi 4

Tried changing the url in .config/octodash/config.json. OctoPi.local, localhost, 127.0.0.1....Port 80 and port 5000 for each of the hostnames, still has the same problem. See the attached screenshot.

[image]

UnchartedBull, 2021-01-02:
Please append /api/ to the end of the host.

UnchartedBull, 2021-01-02:
Hmm that is weird. DLP is still enabled and running?

MrEdskii, 2021-01-02:
[cid:image001.png@01D6E082.0E6F6500]

Yup. Looks like everything is running properly underneath the error. I can switch screens and click buttons, but when I touch the error to clear it, it comes back……takes about 2 seconds and its back.

MrEdskii, 2021-01-02:
Do you think it may be the version of Marlin?

My ender 5 has the creality 4.2.2 main board but I think the marlin version is 1.0.1. At least that s what it shows on the ender display.

MrEdskii, 2021-01-02:
Just an update……

I got another SD card. Reinstalled OctoPi, and Octodash. Configured the Octodash config file to point to url "http://127.0.0.1:5000/api/",

Same thing. Boots properly, Octodash loads. Same error pops up constantly, but Octodash is working properly in the background (moving through different screens and moving axis etc.).

Same versions as I listed in the trouble ticket.

Thanks

Ed

k0riz0n, 2021-01-03:
I ran into the same problem after first updating to 2.1.2, the error was still there.
Yesterday i ran the install script again and now the error is gone... Maybe you give it a try.
`bash <(wget -qO- https://github.com/jneilliii/OctoDash/raw/master/scripts/install.sh)`

k0riz0n, 2021-01-03:
I get the error too, still after updating to 2.1.2, i remember another thread where its mentioned to fix in version 2.2
until then there is a fork without this problem.
` bash <(wget -qO- https://github.com/jneilliii/OctoDash/raw/master/scripts/install.sh)`

MrEdskii, 2021-01-03:
Yes,  this one works. Will wait until 2.2 is out.

Thanks

Ed.

UnchartedBull, 2021-01-04:
2.2 probably won't fix this, more like 2.3 or 3.0. There was some stuff that came in which needs to be released beforehand. Please use jneilliii's fork until then. The error still is really weird. It works just fine for me (everything on latest) plus the 404 indicates that the API just isn't there, which would indicate that DLP isn't installed, but it is according to you. 

I also didn't find any issues over on Ollis Git and can't reproduce this on my side. So, unfortunately, there is nothing I can really do here for now. I'm closing this here (although not fixed), but with a workaround :)

closed 2021-01-04 by UnchartedBull