UnchartedBull/OctoDash #1279: How to view 3D model/preview of G-code like in screenshots?

scottmudge, 2021-01-05:
**What doesn't work?**
In this screenshot, the bency g-code can be previewed with a 3D-model:

https://raw.githubusercontent.com/TimonGaebelein/OctoprintDash/master/screenshots/file_details.png

However, I cannot seem to produce the same effect on my installation, and I cannot find any setting to enable this. How do I do this? Or is it just a fake part of the screenshot?

Latest version of OctoDash (v2.1.2) and OctoPrint, on an Rpi 4.

SwiftingSpeed, 2021-01-05:
You need the Cura thumbnail plugin in Octoprint: https://plugins.octoprint.org/plugins/UltimakerFormatPackage/
And using Cura as a slicer, you just use the "Print with octoprint" button and it happens automatically.

xunleii, 2021-01-05:
If you use PrusaSlicer or SuperSlicer as a slicer, you can use https://plugins.octoprint.org/plugins/prusaslicerthumbnails/. See the plugin documentation to know how to enable thumbnail feature.

scottmudge, 2021-01-05:
Thank you, I got it working with PrusaSlicer after compiling 2.3.0.

closed 2021-01-05 by scottmudge