UnchartedBull/OctoDash #1281: Can't get Octodash to install on fresh Octopi install

KS-Husker, 2021-01-05:
I've gone through this process about 8 times between my 3 printers so far and octodash installed just fine every time. But today I had an issue. I put Octopi on my raspberry Pi 4, then installed Klipper, then went to install octodash and it won't take for some reason. Octoprint works fine with the printer and the klipper configuration. It's the same as on my other printer with the same exact setup that is working perfect with Octodash.

After trying to run the install script, I get these errors below:
bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)

/dev/fd/63: line 2: $'\r': command not found
/dev/fd/63: line 5: $'\r': command not found
/dev/fd/63: line 7: $'\r': command not found
/dev/fd/63: line 9: $'\r': command not found
/dev/fd/63: line 16: $'\r': command not found
/dev/fd/63: line 19: $'\r': command not found
/dev/fd/63: line 27: $'\r': command not found
: invalid option 30: set: -
set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
/dev/fd/63: line 31: $'\r': command not found
/dev/fd/63: line 35: $'\r': command not found
/dev/fd/63: line 47: $'\r': command not found
/dev/fd/63: line 48: syntax error near unexpected token `$'{\r''
'dev/fd/63: line 48: `print() {

BEE-KA, 2021-01-05:
I have the same issue

KS-Husker, 2021-01-05:
Well I'm glad to know I'm not the only one. Thanks.

ssavickiy, 2021-01-06:
I have the same issue

ssavickiy, 2021-01-06:
solved by converting the line break format via Notepad++

superjojo29, 2021-01-06:
> solved by converting the line break format via Notepad++

mind showing me how to do it please?

UnchartedBull, 2021-01-06:
https://github.com/UnchartedBull/OctoDash/commit/507ff71bebe47161ffb0f880304591a80704878d should've fixed this. Please check again and report back :)

superjojo29, 2021-01-06:
> [507ff71](https://github.com/UnchartedBull/OctoDash/commit/507ff71bebe47161ffb0f880304591a80704878d) should've fixed this. Please check again and report back :)

yep all good now! Thank you!

KS-Husker, 2021-01-06:
Yep, You fixed it!! 

Thank you, and thanks for the super fast response!! You guys are awesome!!

closed 2021-01-06 by UnchartedBull