UnchartedBull/OctoDash #1283: The Install script is not working

superjojo29, 2021-01-06:
I get this error after entering the install command:

pi@octopi:~ $ bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)
/dev/fd/63: line 2: $'\r': command not found
/dev/fd/63: line 5: $'\r': command not found
/dev/fd/63: line 7: $'\r': command not found
/dev/fd/63: line 9: $'\r': command not found
/dev/fd/63: line 16: $'\r': command not found
/dev/fd/63: line 19: $'\r': command not found
/dev/fd/63: line 27: $'\r': command not found
: invalid option 30: set: -
set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
/dev/fd/63: line 31: $'\r': command not found
/dev/fd/63: line 35: $'\r': command not found
/dev/fd/63: line 47: $'\r': command not found
/dev/fd/63: line 48: syntax error near unexpected token `$'{\r''
'dev/fd/63: line 48: `print() {

drakkar20d, 2021-01-06:
yes I get the same error

keqinn, 2021-01-06:
I get too.

swissbear, 2021-01-06:
Ah, so here we go, same error here too. And we were sitting around the new bought screen and trying to figure out what we were doing wrong...

FrankX0, 2021-01-06:
Apparently DOS-style CRLF line endings have been added, which cannot be handled by bash.
See https://askubuntu.com/questions/966488/how-do-i-fix-r-command-not-found-errors-running-bash-scripts-in-wsl
@UnchartedBull can you fix this?

UnchartedBull, 2021-01-06:
https://github.com/UnchartedBull/OctoDash/commit/507ff71bebe47161ffb0f880304591a80704878d should've fixed this. Please check again and report back :)

UnchartedBull, 2021-01-06:
Merging with #1281 here. Please post all updates over there.

closed 2021-01-06 by UnchartedBull