UnchartedBull/OctoDash #1291: Switch Filament in Filament Manager _WITHOUT_ sending g-code to printer?

scottmudge, 2021-01-08:
OctoDash tends to cause my extruder to jam and click when I use it to switch filaments, because it's trying to send extra extrude g-code to my Prusa MK3S, when the Prusa does it automatically when it receives a load filament command. So when it should be purging the nozzle, OctoDash is trying to shove through the filament to load it instead, causing horrible noises in the extruder.

I find the built-in menu on my Prusa for loading filament a lot easier to use and more predictable, but I like to use the Filament Manager in OctoPrint to keep track of how much filament I have left. 

Is it possible to configure OctoDash to simply switch what filament is active inside OctoPrint's filament manager when using the filament selector, rather than going through the whole process of actually loading the filament?

UnchartedBull, 2021-01-08:
Turn on Use M600 for filament change in the settings. This will just heat the nozzle and will then send M600 to the printer. The next version will also contain a fast-track (hold on a filament for 3 seconds and it will directly set the new filament, no heating, no other commands)

closed 2021-01-08 by UnchartedBull

scottmudge, 2021-01-08:
Great, thanks!