UnchartedBull/OctoDash #1293: Babystepping settings arent stored permanently

ChristianH99, 2021-01-09:
The save button doesn't execute M500 command to store the offset. It is only stored until the printer is restarted, so the save button has no use.

Steps to reproduce the behavior:
1. Start print
2. Go to adjust
3. Select babystepping
4. Adjust settings
5. Go back
6. Click save

**Expected behavior**
The adjusted offset is stored permanently using the M500 command.

**General Information:**
 - Raspberry Pi 3 B +
 - OctoDash V2.1.2

UnchartedBull, 2021-01-23:
That currently is by design. Maybe I understand the functionality wrong here, but AFAIK Marlin doesn't save babystepping during printer either, but you rather have to send M500, or am I wrong here?

ChristianH99, 2021-01-23:
I‘m not sure how marlin deals with this and I have disabled all that functions on my printer to save memory, but as far as I remember you have to click save to store the value permanently as well.

UnchartedBull, 2021-01-23:
Oh I didn't read your last step properly. I'll probably include a separate safe button in the Babystepping screen. Using the save button on the adjust screen could lead to some problems (since you need to press that to save the adjustments).

ChristianH99, 2021-01-23:
Yes that would be perfect

PR #1621 (2021-03-29): save babystepZ to eeprom
send M500 after setting babysetpZ to save the setting across printer reboots

Fixes #1293
closed 2021-03-29 by UnchartedBull