UnchartedBull/OctoDash #1295: Filament, heating, etc controls

McPrince96, 2021-01-10:
1.  Filament controls: 
     Add extrude and retract controls to the filament menu with 1, 5, 10, 50mm and speed options

2.  Filament preheat temperature presets:
     Something like my drawing. Presets configurable in setting or import from octoprint.
     Preheat button behind the "control" button can possibly go after this change.
[image]

3. Manual bed leveling:
   In Octoscreen there was an option to manually level the bed. It was a picture of a bed and if you pressed on one of the corners 
   it moved to that corner. 
   ex.: Symbol manually level (pressed) --> open picture of bed fullscreen --> press one of the corners --> printhead moves to that 
   corner and bed moves to nozzle --> press other corner --> bed moves down --> head moves to other corner --> bed moves 
   up --> etc.

4. Multiple hotends (weird it's not supported or am i missing something):
    With an option to select wich one to heat up or better yet add an extra symbol in homescreen.
    Fan control can go behind the "control" button to make place imo.

5. Onscreen keyboard/numpad in settings:
    Or maybe an option to change the settings through octoprint in browser. I was pretty bummed i had to connect a keyboard to 
    change even a simple number.

7. (Optional) Terminal support:
    Something like on the bigtreetech displays (but maybe a bit better looking ;) )

8. Multiple fan support:
    I have 2 fans connected to my controller/motherboard. I would be nice to be able to control them both from octodash

9. Individual Z homing options:
    Option to home Z individually. Maybe with a house symbol between the move arrows. One for XY and one for Z

10: Reset button:
      Don't know if this is the same as the reload or kill button but i i press the reload or kill button it doesn't reset my printer, only 
     the raspberry.

11. Disable stepper motors:
      Maybe this can go in a menu together with the terminal, marked as 'advanced controls'

12. Layout idea
      In my opînion the 'filament' button on the homescreen can move behind the 'control' button.
      This way the fan button can stay and there will be enough place for a second hotend symbol.

13. Bltouch controls
      Maybe also behind an 'advanced controls' button

I hope a lot of these features (maybe even all of them) will make their entrance in a future update.

UnchartedBull, 2021-01-23:
Those are a lot of good suggestions! I'll make sure to work myself through those some time, can't guarantee a release date for any though.

thegarbz, 2021-01-31:
@McPrince96 there's some good stuff there. Regarding you suggestion 11. You can do that with the custom button functions already. There's a wiki guide https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions and you can set one of the buttons to issue command "M18". There's even a suitable icon available :) https://fontawesome.com/icons/cogs?style=solid

salocain, 2021-04-13:
multiple hotends ,please !!!!

a good lot of ideas

pciavald, 2021-04-14:
I think these presets should be read from filament manager

htrabbit, 2021-10-30:
these are great ideas, if i could only choose a few.......
i really would like to see 1,2,5 & 8.