UnchartedBull/OctoDash #1298: Error when installing AMD x86 _64 package

bdog1, 2021-01-12:
Attempted install of OctoDash on Linux Bionic Beaver and received the following error relating to no repository found for dependencies.
Redirecting output to ‘wget-log.1’.
Installing OctoDash v2.1.2, x86_64
Installing Dependencies ...
E: The repository 'http://ppa.launchpad.net/pmjdebruijn/avrdude-release/ubuntu b                                                                                                                                                             ionic Release' does not have a Release file.
W: Skipping acquire of configured file 'stablenightly/binary-amd64/Packages' as                                                                                                                                                              repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't h                                                                                                                                                             ave the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/i18n/Translation-en' as re                                                                                                                                                             pository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't hav                                                                                                                                                             e the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/dep11/Components-amd64.yml                                                                                                                                                             ' as repository 'https://download.docker.com/linux/ubuntu bionic InRelease' does                                                                                                                                                             n't have the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/dep11/icons-48x48.tar' as                                                                                                                                                              repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't h                                                                                                                                                             ave the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/dep11/icons-64x64.tar' as                                                                                                                                                              repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't h                                                                                                                                                             ave the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/cnf/Commands-amd64' as rep                                                                                                                                                             ository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't have                                                                                                                                                              the component 'stablenightly' (component misspelt in sources.list?)
bdog@octolinux:~$ bash <(wget -qO- https://github.com/M1dn1ghtN1nj4/OctoDash_Bui                                                                                                                                                             lds_UNOFFICIAL/releases/downloa^C
bdog@octolinux:~$ bash <(wget -qO- https://github.com/M1dn1ghtN1nj4/OctoDash_Bui                                                                                                                                                             lds_UNOFFICIAL/releases/download/2.1.1-110820/install-amd64.sh)

UnchartedBull, 2021-01-12:
This seems like a problem with your package manager, rather than with OctoDash. Please execute `sudo apt update && sudo apt upgrade` and check whether those yield any errors.

bdog1, 2021-01-12:
Thank you for taking the time to reply!

As a matter of fact I do recieve a similar message when I issue a sudo apt update. "W: Skipping acquire of configured file 'stablenightly/binary-amd64/Packages' as repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't have the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/i18n/Translation-en' as repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't have the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/dep11/Components-amd64.yml' as repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't have the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/dep11/icons-48x48.tar' as repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't have the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/dep11/icons-64x64.tar' as repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't have the component 'stablenightly' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'stablenightly/cnf/Commands-amd64' as repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't have the component 'stablenightly' (component misspelt in sources.list?)
E: The repository 'http://ppa.launchpad.net/pmjdebruijn/avrdude-release/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details."

Any idea how to go about resolving my potential package manager issue?

Bruce

bdog1, 2021-01-12:
Im just going to upgrade to the latest version of Ubuntu -Focal Fossa version 20.04 and see if that brings resolution. I will update this posting with results.

closed 2021-02-02 by stale[bot]