UnchartedBull/OctoDash #1304: Can't retrieve layer progress

JohanB55, 2021-01-13:
Hi,
I have installed octoDash on my raspberry but now I get a Warnig

Http failure response for http://octopi.local/plugin/DisplayLayerProgress/values: 404 not found

I have looked at the internet but i can not find a solution.
Can you help me?

[image]

jonnalexander, 2021-01-15:
Im wracking my brain to figure this out too. Tried looking through other reports and nothing ive seen solves my problem, ive tried reinstalling too.

Sekisback, 2021-01-15:
same issue on my side

Silentee, 2021-01-16:
Are you sure you have the DisplayLayerProgress plugin installed? Without additional info such as version numbers (octoprint, octopi, octodash), slicer used, and what settings you have setup in the DisplayLayerProgress plugin then it'll be hard for anyone to help you.

Sekisback, 2021-01-16:
In my case, I update to 2.1.2 and had an issue with the sleepfunction, so i downgraded back to 2.0.
Octoprint Version 1.5.2
Octopi 0.17.0
Raspberry Pi 3 Model B Rev 1.2
DisplayLayerProgress 1.24.0
SlicerCura 4.8

AtomicRas, 2021-01-17:
I tried switching my port in OctoDash from 80 to 500 like someone said they did to get working. Now it wont connect just hangs. How can i switch the port now?

I uninstalled and reinstalled but i think it is still keeping the port 500 from before.

ammas09, 2021-01-17:
I also see this occasionally when the printer is connected but not printing.  I definitely have the DisplayLayerProgress plugin installed.  If I can help by providing more info or testing anything let me know.

JohanB55, 2021-01-17:
I have the DisplayLayerProgress plugin installed and it works great now! 
Everybody thanks fot the help!!

ammas09, 2021-01-17:
> I have the DisplayLayerProgress plugin installed and it works great now!
> Everybody thanks fot the help!!

How did you get it working?  I still get the error and I have the plugin installed already (I installed it during the OctoDash install propmts)

UnchartedBull, 2021-01-23:
Reopening since the error does seem to pop up again. The workaround for now is to use @jneilliii's custom fork with DisplayLayerProgress disabled: https://github.com/jneilliii/OctoDash.

If anyone who has the problem could please reply with the following information:

- DLP version
- OctoDash version
- Did you print a file that was uploaded BEFORE the installation of DLP
- Does the OctoPrint WebUI correctly show the Layer Progress?

Cyberbastler, 2021-01-23:
What doesn't work?
I often get the "Can't retrieve layer progress!" / "Http failure response for ... values: 0 Unknown Error"
Would guess approx. every 30 minutes

What did you already try?
Updated everything, checked configuration. I even updated the OS directly; executed "apt update/upgrade", etc.

General Information:
Raspberry Pi 4 (4 GB variant)
OctoPi 0.17.0
OctoPrint 1.5.3
Latest OctoDash-version (installed using the script)

Cyberbastler, 2021-01-23:
Changed from octopi.local to localhost in config. May have solved the issue.

Cyberbastler, 2021-01-24:
As of right now I would that's it in my case: Use "localhost" and the problem seems to be gone!

UnchartedBull, 2021-01-25:
There seems to be something really fishy with octoprints mdns atm. Lots of people reporting problems which go away once localhost is used.

jneilliii, 2021-01-26:
It might have something to do with IPv6. I do know that there were some changes to mdns in one of the recent releases.

tubbybtch, 2021-01-26:
I'm persistently having the same error: "Can't retrieve layer progress - Http failure response for http://localhost/plugin/DisplayLayerProgress/value - 404 NOT FOUND"

I've tried changing url in ~/.config/octodash/config.json.  Either I get no connection or the above error.  I have tried network alias, url and localhost.

Three weeks getting this far...really frustrated!  (and I still don't have touchscreen working)

General Information:
Raspberry Pi 4 (4 GB variant)
OctoPi 0.18.0
OctoPrint 1.5.3
Latest OctoDash-version (installed using the script) 
Hyperpixel4

Mattch788, 2021-01-28:
i an getting the same error but on the octoprint screen rather than on octodash, i also am unable to update octoprint now that octodash is installed for some reason, has anyone had this same iisue?

UnchartedBull, 2021-01-29:
Ok I still really don't know whats going on here, but here are a few things to try so we can narrow this down:

- Please check if you're also getting an error in the WebUI as @Mattch788 mentioned
- Did you get the low voltage warning in OctoPrint / how is the load on the Pi (screenshot of `htop` or average percentage)
- Does the error go away if you change the polling interval?
  - `nano ~/.config/octodash/config.json`
  - Go all the way down to `pollingInterval` and change that to 5000
  - Reboot
 
After this change OctoDash will only ask OctoPrint for new data every 5 seconds, if this error is somehow load related this should get rid of the error.

Cyberbastler, 2021-01-30:
> As of right now I would say that's it in my case: Use "localhost" and the problem seems to be gone!

It came again when I cahnged to octopi.local and went when I cahnged back. So this it was for me.

jkuusama, 2021-03-15:
Latest everything, Pi 3B+. I had the same, on Octodash screen only, and only when idle. Yes, I did print some files before installing Octodash and DLP. When printing, everything works as expected. Seems to have gone away after switching to localhost. (Fingers crossed, but I haven't seen it after I made the change about 30 mins ago). Thank you @Cyberbastler !

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550
closed 2021-03-18 by UnchartedBull