UnchartedBull/OctoDash #1310: Screen Black with Dash Top Left - Tries to show OctoDash

spankurmonkey, 2021-01-14:
**What doesn't work?**
Does not bring up Octodash - screen flickers for a minute and can see the screen but then it goes back to black screen with dash at top left.  I have VNC viewer enabled and itermittently on the viewer I can see the Hey there! but on the physical screen it just shows black with that dash other times after rebooting VNC flickers like its going to show it but then only shows the black screen.

**What did you already try?**
set the console auto login

**General Information:**
 - RPI 3 Model B - Inno-Maker 3.5 320x480
 - OS Info current version OctoPi Download from Github
 -  OctoDash Version current 2.1.2
 - OctoPrint Version 0.17.0

**Additional context**
videos of it both physical and via vnc
https://www.youtube.com/watch?v=SKYKnRhBgL0
https://www.youtube.com/watch?v=HJeFxbkoBFo

spankurmonkey, 2021-01-14:
Ok after hours of trying everything since I know nothing I tried the first step in this issue:

https://github.com/UnchartedBull/OctoDash/issues/470

It worked... now its showing and so far the screen seems responsive.

However, the only issue I am having is when it first boots up the screen is at full resolution but then its not.... not sure what the problem is.

VNC shows it full but the actual screen it's not  ( https://www.youtube.com/watch?v=4JhXVO2JeBk )

[image]
[image]

AtomicRas, 2021-01-17:
I've been trying to get over this DLP error almost my entire weekend. When you switch options within OctoDash on the touch screen, how can one reset those options back to default?

Because it was at least loading up the OctoDash GUI not it just hangs on "Connecting"

UnchartedBull, 2021-01-23:
@AtomicRas https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#manually-adjusting-the-settings, default settings can be found here: https://github.com/UnchartedBull/OctoDash/blob/master/src/app/config/config.default.ts. You can also always just remove that file and start over again.

@spankurmonkey you probably need to look into overscanning to fix the screen borders.

closed 2021-01-23 by UnchartedBull

spankurmonkey, 2021-01-23:
I finally got it working....  but how it was all hap hazardly done.... the instructions for the InnoMaker 3.5 touchscreen are far from clear and accurate.