UnchartedBull/OctoDash #1317: More Display Options from DisplayLayerProgress

Rneil-ca, 2021-01-16:
Hi,

First of all have to say I am really happy with OctoDash, it works really well for my purposes.  DisplayLayerProgress has a couple of things that it can show you that I find really useful, it would be nice to have the option to have these displayed on the Octodash screen as well:

estimatedEndTime,
changeFilamentCount3,
changeFilamentTimeLeft,
 estimatedChangedFilamentTime

Thanks,
Ryan

AtomicRas, 2021-01-17:
I'm getting 502 Bad Getway errors from the DLP plugin.

Pointing to this url http://octopi.local/plugin/DisplayLayerProgress/values

UnchartedBull, 2021-01-23:
@Rneil-ca those features are mainly used for a dual extruder setup, right?

@AtomicRas 502 Bad Gateway means that something with your internet configuration is really messed up. OctoDash doesn't have access to any of the configurations, so there is nothing I can do from OctoDash side. Does everything else work?

Rneil-ca, 2021-01-24:
Hi UnchartedBull,

No they are not used for dual extruders. If you are printing multicolor prints with a single extruder, it will tell you the time left  until you have to change the filament(changeFilamentTimeLeft).   It looks for the M600s in the gcode and gives you a time estimate till it reaches that point.  It will also tell you the number of filament changes that you need to do for this print (changeFilamentCount) and the estimate actual time that the next change will happen(estimatedChangedFilamentTime).

The estimatedEndTime option, tells you what actual time will be when the print is completed.  Saves you from doing the math in your head.

Not a critical need for these but they would be a nice option to be able to display this info on the screen.

Thanks again for all  your hard work on this.

Ryan

UnchartedBull, 2021-01-25:
Gotcha. Probably will be something for a popup (i.e. by clicking on the layer progress or similar) to not bloat up the main UI too much.

Rneil-ca, 2021-01-26:
I agree, that would be a great way of doing it.

Thanks,
Ryan

thegarbz, 2021-01-31:
@Rneil-ca as a matter of interest why do you need estimatedEndTime from DLP? What math are you doing in your head? Octoprint / Octodash already display an ETA to the end of the print without the need to use DLP. This has other benefits because by reading the ETA from Octoprint itself you can get the time controlled by various plugins (e.g. print time genius, or the M73 ETA Override plugin). Getting it from DLP would bypass all plugins that "optimise" this ETA in Octoprint.

Rneil-ca, 2021-01-31:
Hi thegarbz,

The estimatedEndTime shows you the actual time that the print will be done (eg. 8:55pm). vs number of hours/minutes left to print. Makes it easier to know when a longer print will be done.  Like I mentioned not a big thing, just something that is nice.

thegarbz, 2021-02-01:
*Facepalm* sorry I was thinking back to memory and I thought that was already on the display. But I just fired up the printer and I got it confused with the elapsed time, not the ETA time. Don't mind me :)