UnchartedBull/OctoDash #1326: Display Layer error

Mattch788, 2021-01-22:
**What doesn't work?**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I continue to get the error can't retrieve layer progress. 

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.
I have made sure the layer plugin is activated during the install. 

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]
Using raspberry pi 3b+
Ictoprint v1. 4.0

**Additional context**
Add any other context or screenshots about the feature request here.

Zini72, 2021-01-23:
I often get the "Can't retrieve layer progress!" / "Http failure response for ... values: 0 Unknown Error"

I would guess approx. every 30 minutes

My HW/SW:
Raspberry Pi 4 (4 GB variant)
OctoPi 0.17.0
OctoPrint 1.5.3
Latest OctoDash-version (installed using the script)
In addition, I updated the OS directly; executed "apt update/upgrade", etc.

UnchartedBull, 2021-01-23:
There is absolutely no information in the original issue. Closing as duplicate of #1134.

@Cyberbastler could you head over to the other issue and provide the details requested there? This bug is currently unreproducible for me, so I need as much information as possible. Thanks!

closed 2021-01-23 by UnchartedBull