UnchartedBull/OctoDash #1334: Cant install Octodash with given URL

berlinchris81, 2021-01-23:
**What doesn't work?**
Install of Octodash.

**What did you already try?**
Check URL twice. 

**General Information:**
 - Hardware Raspberry Pi 3
 - OS Info Raspbian
 - OctoDash Version  none
 - OctoPrint Version   v1.5.3

**Additional context**
This Command ---- bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh) ---- does not work - given URL does not exist.

Pls Help

Greetings from Berlin 

Chris

b1u3git, 2021-01-24:
I tracked down the script here:
https://raw.githubusercontent.com/UnchartedBull/OctoDash/main/scripts/install.sh

Installing now myself. 

Please let me know if this is not kosher with guthub to have done so?

berlinchris81, 2021-01-24:
Thats works fine ! Thank you!

closed 2021-01-24 by berlinchris81