UnchartedBull/OctoDash #1335: Unable to install OctoDash on new OctoPi system using install script

jordancoleman, 2021-01-24:
Have a fresh OctoPi system running on a Pi 4, and trying to install OctoDash via `bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)` in an `ssh` session. Running that command verbatim (as pasted from the `README.md`) fails silently and returns to a prompt. After some investigation, it appears the URL referenced in the provided command leads to a 404 page on GitHub, and so the command tries to execute the error page.

I altered the command to reference `https://raw.githubusercontent.com/UnchartedBull/OctoDash/master/scripts/install.sh`, which at least provides code to execute. However, the script fails with `stty: 'standard input': Inappropriate ioctl for device` when run.

After some investigation, it appears the script fails at the `stty -echo` in the `_checkbox_input()` function.

b1u3git, 2021-01-24:
Same here - Fresh OctoPi 4, found same alternate script path.  

Install appears to have completed for me.   Went thru handful of config options, including option to reboot at script end.   Doesn't appear to be auto-loading OctoDash at startup however.  

Not sure how much this helps, but there it is.

UnchartedBull, 2021-01-24:
Thanks @b1u3git. I just updated the master branch to main branch yesterday und forgot to update the README. I now updated the README with the new branch name so everything should work again :)

closed 2021-01-24 by UnchartedBull

jordancoleman, 2021-01-24:
Sorry, but the original issue that led to opening this issue still exists. I think @b1u3git had a slightly different problem.

Right now, when I run `bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)` per the (current) `README.md`, the script just stops in the middle and returns to the shell prompt -- now without any error messages, but nothing happens.

Output below:

<pre>
pi@octopi:~ $ bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
Installing OctoDash v2.1.2, armv7l
Installing Dependencies ...
[sudo] password for pi: 
All packages are up to date.
bc is already the newest version (1.07.1-2).
desktop-file-utils is already the newest version (0.23-4).
libappindicator3-1 is already the newest version (0.4.92-7).
libatspi2.0-0 is already the newest version (2.30.0-7).
libavahi-compat-libdnssd1 is already the newest version (0.7-4+b1).
libnotify4 is already the newest version (0.7.7-4).
libnss3 is already the newest version (2:3.42.1-1+deb10u3).
libsecret-1-0 is already the newest version (0.18.7-1).
libuuid1 is already the newest version (2.33.1-0.1).
libx11-dev is already the newest version (2:1.6.7-1+deb10u1).
libxss1 is already the newest version (1:1.2.3-1).
libxtst6 is already the newest version (2:1.2.3-1).
ratpoison is already the newest version (1.4.9-1).
xinit is already the newest version (1.4.0-1).
xserver-xorg is already the newest version (1:7.7+19+b8).
libgtk-3-0 is already the newest version (3.24.5-1+rpt2).
libpam0g-dev is already the newest version (1.3.1-5+rpt2).
x11-xserver-utils is already the newest version (7.7+8+rpi1).
xdg-utils is already the newest version (1.1.3-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@octopi:~ $ 
</pre>

jordancoleman, 2021-01-24:
Additional info: When the script exits (prematurely, as in the example above), it leaves the terminal in a non-standard state. It appears the `stty -echo` in the script executed before it exited and the terminal echo was never reinstated before returning to the prompt.

jordancoleman, 2021-01-24:
Replacing calls to `tput` with a noop (by adding `tput() { echo -n }` at the top of the script) allows the script to run, provided one is willing to decypher unformatted output from the checkbox UI, which no longer works properly without `tput` (it accepts input, but the terminal output is all garbled).

I've now got OctoDash running, but still not sure why the install script is broken.

jordancoleman, 2021-01-25:
It occurs to me in retrospect that this issue is likely due to terminal differences. The install script's reliance on advanced terminal features means it may not behave consistently depending on how you're connected to the Pi. I'm using `ssh` from Mac Terminal, which shows up as terminal type `xterm-color`, which is likely different from the terminal types used by various Linux systems' terminal apps and from the Pi console.

To test this, I tried running the install script using the one-liner from the `README.md` from inside `screen` (which has a terminal emulation as part of its functioning), and the script ran fine. So for people having the same problem I did, who are Googling around looking for a fix, hopefully you'll find this and see that the workaround is to run the script inside `screen`.

If the install script only supports certain terminal types, that should be documented somewhere. It would be better, of course, to have a mode for the install script that does not require interactive keyboarding to select install options.

UnchartedBull, 2021-01-25:
Mhh that certainly is really weird. I'm also SSHing into the Pi from my Mac and I don't have any problems with the script. Thanks for the info! I'll make sure to look around for an alternative which does run on more systems. If you have any ideas let me know :)

jordancoleman, 2021-01-26:
Interesting that I have this problem and you don't. Would be good to know what terminal type `echo $TERM` returns on both your local Mac shell and in an `ssh` session to your Pi.

In my case, both return `xterm-color`. The Mac is running Catalina 10.15.7. On my Pi, I'm running a fresh and relatively clean install of the latest OctoPi image (`2019-09-26-octopi-buster-lite-0.17.0.img`), though I did an `apt update && apt full-upgrade` on it after installing, and also attempted to install the TouchUI autostart stuff (before running into problems and deciding to try OctoDash -- which is great, BTW).

UnchartedBull, 2021-01-29:
[image]

Does seem to be almost the same. Is there a difference between xterm-color and xterm-256color?

Mojofreem, 2021-02-27:
Can confirm there are terminal specific commands that kill the installer. I have my terminal set to `VT-102`, and I was not even seeing _any_ output, simply a silent failure, with status code 1 (ie., `echo $?`). I changed my terminal mode to `xterm-256color`, and the install worked as advertised. I suggest adding a disclaimer in the installation instructions about requiring a specific terminal emulation mode (`xterm-256color`), or, removing the terminal specific coding and replacing with a common denominator subset of codes (the `ANSI` subset should be reasonably safe).

UnchartedBull, 2021-03-02:
Ok I have absolutely no idea about terminal emulation modes and unsupported stuff in them. I usually just use zhs and leave it at that. 

All the fancy selection stuff is coming from Inquirer (https://github.com/kahkhang/Inquirer.sh). If you have any ideas on how to make this more compatible please let me know. Otherwise I'll just add the note regarding the `xterm-256color`. Thanks!