UnchartedBull/OctoDash #1336: Unable to tap/press and scroll in files listing and custom action buttons

matrixvibe, 2021-01-24:
**What doesn't work?**
I started noticing the inability to scroll through the custom action buttons when I added an extra one and couldn't get them to scroll up and down. Then I tried to push and scroll in the files screen and it didn't work either (the scroll bar works fine).

**What did you already try?**
I've attempted to look for any calibration or touch settings but I'm relatively new to the Pi world so I don't know what else to try other than calibration. Haven't actually done it just yet.

**General Information:**
Just got a Raspberry Pi 4B and a touchscreen a couple of days ago. New to Octoprint and Octodash. The touchscreen is a resistive touch model, not sure if this is the reason or there's any further configuration that needs to be enabled for scrolling to work.

 - Hardware: Raspberry Pi 4 Model B Rev 1.2 - 4GB Ram
 -Touchscreen: Miuzei (https://www.amazon.ca/Miuzei-Raspberry-Display-Full-Angle-Raspbian/dp/B07YDBFY8F), which appears to be the 4inch HDMI Display-C (http://www.lcdwiki.com/4inch_HDMI_Display-C) as the installation commands were the same.
 - OS Info: Octopi Version 0.17.0
 - OctoDash Version 2.1.2
 - OctoPrint Version 1.5.3

**Additional context**
Octodash is looking great so far. 

Thanks in advance!

UnchartedBull, 2021-01-25:
That's most likely a hardware limitation of your screen. Resistive touchscreens are know to be really bad with gestures, so unless there is some fancy stuff you can enable in the drivers you need to use the scrollbars on the side I think. I enabled all the touch-controls I can on the OctoDash side.

matrixvibe, 2021-01-26:
> That's most likely a hardware limitation of your screen. Resistive touchscreens are know to be really bad with gestures, so unless there is some fancy stuff you can enable in the drivers you need to [...]

Thanks for the reply. The driver doesn't look too hopeful. Perhaps I'll send this back and find a capacitive one instead.

Any touchscreen you'd recommend?

Thanks!

matrixvibe, 2021-01-28:
Ended up getting a 7 inch HDMI capacitive touch display that interfaces via USB. The scrolling works beautifully now!

closed 2021-01-28 by matrixvibe