UnchartedBull/OctoDash #1340: Sub folder on SD card

Jeggo65, 2021-01-24:
**Describe the bug**
I am not able to print any files from subfolder on the SD card in my Prusa i3 MK3S.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to SD card
2. Click on any file located in a subfolder
3. Error: File not found

**General Information:**
 - Hardware  Raspberry Pi 4B with 4GB
 - OS Info OctoPi 0.18
 - OctoDash Version 2.1.2 (latest stable)
 - OctoPrint Version 1.5.3

UnchartedBull, 2021-01-29:
I'll have a look into this. Printing atm, so might be a while. Can you print fine from the SD Card via the OctoPrint web interface?

Jeggo65, 2021-02-02:
From OctoPrint a subfolder SD card print is working. Printing from the root directory works. But I also never had a successful upload to the local SD card via OctoPrint.
If I need to print a large number of the same parts, I will upload the print files to the SD card on an PC.

vlaine, 2021-03-02:
I am having the same issue using the steps listed by the author.  
Printer: Prusa Mk3s+. 

Also, if I use my printer LCD (not Octodash) to print a file located on a folder on my SD Card, OctoDash keeps popping the error "Can't retrieve job status".

UnchartedBull, 2021-03-11:
Ok I just had a look at this and I'm a bit puzzled. Files in subfolders look really weird (DOS 8.3 format) and are displayed in a single directory (both in OctoDash & OctoPrint). This is known according to this: https://community.octoprint.org/t/why-does-octoprint-display-files-on-the-printers-sd-so-weirdly/2411. How have you managed to get sub-folders working on the SD-Card?

Also I can select files from the SD-Card just fine. The error with the message will be fixed with the next release. There will be a very very limited amount of information available though, since OctoPrint also doesn't really know what's going on if your print from your SD Card.

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550
closed 2021-03-18 by UnchartedBull