UnchartedBull/OctoDash #1346: Remove M18 E from the Filament Change Script

Rneil-ca, 2021-01-25:
Hi,

Is it possible to remove the M18 E from the filament change script?  I love this feature, but unfortunately I am running Klipper and it shuts down all steppers when the M18 E command is issued.  This means if I use this feature to change the filament while paused during the print, I cannot continue the print.

Thanks,
Ryan

Rneil-ca, 2021-01-26:
Looked into this myself a bit, the M18 E is built right into the code itself. I was able to remove this by carefully editing the app.asar file (I would not recommend this).  It now works great.

Would be nice if there was a way to customize the before and after filament change scripts, maybe with a text file that could be edited.  

Ryan

UnchartedBull, 2021-01-29:
Yeah that part unfortunately is hardcoded. I'll make this configurable in one of the following versions.

UnchartedBull, 2021-02-17:
Fixed with #1359.

closed 2021-02-17 by UnchartedBull