UnchartedBull/OctoDash #1347: Touch Input occasionally stops working.

SaKiEQ, 2021-01-27:
**What doesn't work?**
Every day or two, the touch-input stops reacting. Unfortunately not in the same 'software position', but as fas as I can 
tell never in the home screen, rather in a secondary view like 'adjustment' or 'or 'control'.

Ocotprint keeps running under the hood. OcotoDash keeps updating screen information, just any user input does not work.  

**What did you already try?**
All I could do so far is to make sure everything is on the latest software versions. Everything returns to normal after a restart of the Pi, or sometimes if I just let the system sit idle for a while. 
Killing the octodash process and bringing it back with:
service getty@tty1 stop
service getty@tty1 start

Is working, so I am assuming it is not a hardware issue. 
Also I made sure all the connections are secure, beween pi and display etc, so the interface should not be the culprit. 

**General Information:**
RETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
debian_version:   10.7
Linux octopi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

**Rasperi Pi 4B Info:**
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : b03112
Serial          : 10000000673286b2
Model           : Raspberry Pi 4 Model B Rev 1.2

**System Info:**
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: false
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1904373760
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x50000
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin Ver 1.0.2

**Hardware:**
Raspberry Pi 4 Model B
5" Touch Screen -  Waveshare 

**Additional context**
80% of the time it works just fine, all the features, my custom actions work just fine. Until I work with the machine and hit a few too many inputs in too short of a time and the touch becomes unresponsive. Quiet often the situation resolves at the end of a print, when the printer it self goes idle...
One hunch I am having is, that the Octodash is locking touch input until it receives a 'completion' acknowledgment of which ever task was called last, but occasionally that feedback gets lost  ?

Can you point me in  a direction of a error log etc, where I can start troubleshooting ?

SaKiEQ, 2021-01-27:
Where i am stuck today... touch is completely unresponsive, tried attaching mouse and keyboard to try if ESP / Enter / Mouse Click would resolve it, but no such luck. 

[image]

SaKiEQ, 2021-01-27:
I connected via VNC to the Pi and could also not interact with Octodash, while it works just  when the 'touch disconnect' problem does not happen. Under normal operations I can remote control via touch the Octodash UI via VNC, but not so when it shows this issue. 

So I assume its a driver that buggers up, rather then a hardware issue.

UnchartedBull, 2021-01-29:
Please monitor your Pi with `htop` while executing multiple commands to "force" the issue. It could theoretically OctoDash leaking memory (which I'm not aware of and doesn't occur on my system). With htop you'll hopefully find the process that buggers up the system, we can then go from there :)

SaKiEQ, 2021-01-30:
will do ! thanks for the suggestion - will report back when I get some results. 

Sarah Kimberly Euschen
Production Designer
Westbay Lagoon 
Doha, Qatar

https://themagicalworldofsakie.wordpress.com

SaKiEQ, 2021-02-09:
Hi, 

just wanted to give you a quick feedback. Of corse following Murphys Law - nothing happened since then. The touch screen issues where daily before I wrote you, now since then not once - ironic. 
I keep periodically monitoring the tasks. 

The one and only thing that I changed was the power supply to the Pi and Screen. Before it was an official Raspberry wall brick to USBC. Which for nothing more then cleanliness I replaced with a industrial 5V supply that powers other devices within the same printer enclosure. 
I did not really get under voltage errors before, neither now. But logic would tell me that the only thing that changed must be the reason for previous issues and explain the absence now. 

I will let you know if something comes up ! 

Sarah Kimberly Euschen
Production Designer
Westbay Lagoon 
Doha, Qatar

https://themagicalworldofsakie.wordpress.com

UnchartedBull, 2021-02-09:
Could be that the new PSU has a cleaner ground level or is somehow grounded to your homes ground via another component (had a similar issue with the official raspberry touchscreen and ghost touches - an ethernet cable made the behaviour go away since then everything was grounded to the main ground).

Closing this here, if it happens again please reopen!

closed 2021-02-09 by UnchartedBull

SaKiEQ, 2021-02-09:
Thanks for the patience !

yes, grounding can be an issue - not on this machine thought. Had another one, where the printer PSU was detached, but driver board (not OctoPrint or Raspi) was still on chassis, USB ground made contact with the metal frame. 
I could measure voltages up to 80+VDC to ground. No real current, but enough to throw USB and other things off every once in a while. 

On this machine, the Raspi talks to a Arduino for enclosure controls, which was powered separately, printer was powered on its own, Raspi on its own, some LED's on their own. 
So I consolidated all the 5VDC, 12 + 24VDC onto one power supply, which solved some other issues with the Arduino too. 

Despite it being plug-and-play for most part, a bit of engineering is still required ;) 

Sarah Kimberly Euschen
Production Designer
Westbay Lagoon 
Doha, Qatar

https://themagicalworldofsakie.wordpress.com

UnchartedBull, 2021-02-09:
That's half the fun of the 3D printing :)