UnchartedBull/OctoDash #1350: Edit API key after first start up

zdog90210, 2021-01-28:
**What doesn't work?**
I need to change API key after the first boot of octodash. I restored a backup of ocroprint after I got octodash working after nearly 20 hours of trying and now I need to edit the API key to work with the backup I restored.

**What did you already try?**
Reinstalling octodash from nightly 
Using the remove script from the main branch 

**General Information:**
 - PI 4. KUMAN tft GPIO 3.5 in screen. 
 - I am running the nightly build as it's the only way I got octodash working correctly.

zdog90210, 2021-01-29:
I figured out my problem for anyone else searching `sudo nano ~/.config/octodash/config.json`

closed 2021-01-29 by zdog90210

RuffWorksTech, 2022-02-09:
I needed to change the API key as well as the RPi instance name. This helped me get it done very quickly. Thank you!