UnchartedBull/OctoDash #1358: Support for OctoLight

mikulik86, 2021-01-29:
I'd like a support for this plugin: https://github.com/gigibu5/OctoLight in the Custom Actions of the OctoDash. The same way the PSUControl is supported. 

Thanks

UnchartedBull, 2021-01-29:
Leaving this open for discussion.

thegarbz, 2021-01-31:
Only problem I see with this is that there's 10 other plugins for Octopi which also control LEDs and all do it in a different way :-/

mikulik86, 2021-02-01:
> Only problem I see with this is that there's 10 other plugins for Octopi which also control LEDs and all do it in a different way :-/

Just like there is 10 other plugins for controlling printer power besides PSUControl. So what?
I find this one most simple and practical.

thegarbz, 2021-02-01:
Errr, there's not. There's only one plugin that functions as PSU control from the I/O. But that's not my point. 

My point was more that simple may not be an ideal target for a feature. A generic I/O plugin may be better to implement, or a more comprehensive plugin that supports a wider range of use cases than simply toggling the light off and on. e.g. Implementing the same functionality in Octoprint-RGB-Status means you cover the users of all PWM strips, WS281x strips, as well as GPIO users such as yourself.

mikulik86, 2021-02-02:
I meant all the plugins for wireless sockets etc. 
Anyway, if you say, that RGB-Status offers same functionality as OctoLight and more (I wasn't aware of that and wanted to check the plugin out, but the current version doesn't run under latest OctoPi), than it sure would be great.
An integrated easy way to control a light would be surely appreciated.

thegarbz, 2021-02-02:
Yeah it was just a suggestion to the maintainer to implement something that could work for a larger audience. I'd like the idea too. That said if RGB-Status doesn't work on the latest Octoprint the topic may be moot. Last push was 8 months ago. Maybe it's not being maintained :( I'm still a few versions behind on Octoprint.

UnchartedBull, 2021-02-09:
Have you tried achieving the same control with the Enclosure plugin? This on is already integrated and adding the functionality to add toggling a GPIO pin via the custom actions would be really easy and could be done with the next update. Enclosure plugin is also still running on 2.x unfortunately, but the author said the python 3 version is in the making.

SaKiEQ, 2021-02-10:
I followed Timos advice with the Action Commands / Enclosure plugin route, while it works fine, it took a while to figure out the philosophy behind it, since there are no step by step tutorials for noobs and outsiders. 
GCode contains a lighting enclosure code - which is sent/echoed on the printer data stream, picked-up by ActionCommands and turned into an Action. -> LIghts on.off. 

The only thing philosophically I have against this is, there are more things involved then needed for a simple problem. OctoDash->GCODE->OctoPrint->ActionCommands, for the sake of trouble shooting a single step would be 'cleaner' but clearly more work and maintenance. 

While on the topic, the only vexing thing about it is - that once in print/operation mode, the action pallet can no longer be opened - hence lights not directly controlled - nearly defeats installing them on a button in the first place.

UnchartedBull, 2021-02-12:
If you have it working with the Enclosure Plugin I can add support for triggering outputs there directly via Custom Actions so you won't need the ActionsCommands part anymore.

Custom Actions probably will come in the version after the next release.

mikulik86, 2021-02-16:
I have it working with the Enclosure Plugin the whole time, but I take it only as a workaround, since it is quite unpractical.  I didn't find a way to make it toggle, so there must be two actions for ON and OFF. Thus in OctoDash must be also two actions used up. Furthermore this way it doesn't share the current state of the GPIO. If I toggle Octolight off, then turn on using Enclosure Plugin, then I have to click 2x on the Octolight icon to turn off again. It is really not a good solution.

mikulik86, 2021-02-16:
And yes, there should be option to access some Actions during print.

SaKiEQ, 2021-02-17:
Me personally I dont like toggle options, since they do not have  a defined state, specially when activated via push-button logic. I prefer to have to have separate on/off commands, or a read-back / pull function to get the current state. 
This can become a problem when you have more then one source / switching location. 

In my case I switch via web interface when accessing the enclosure camera, via PIR sensor when I walk up to the machine, or via button on the console / touch screen. Hence being able to send clear On/Off is vital in my case. 

I solved most of this by having a dedicated arduino on USB/Serial that receives commands, so I can give it more logic and control. The same controller feeds temp/humidty info back to octopi/dash, controls the fans in the enclosure etc.. 

All the octoprint does is trigger dedicated python scripts that send the commands to serial. 
Octodash just provides the 'trigger' sort of speak, via the push button, since I am giving it its own python script to be executed, so one Action Command button in Octodash toggles the light, while Enlcosure plugin it self has two visible user button for On / Off, the same for the camera script - when I access the camera, it cant toggle (random), it has to go to defined "on" state.

SaKiEQ, 2021-02-17:
You can also send a custom command to octoprint, like:  "command": "M118 A1 action:lightsOff"
In Octoprint to 'intercept' the action in ActionCommands and trigger for instance a python script that 
drives the GPIO/Lights or which ever way you have integrated your lights. 
In the python script you can implement which ever logic you require to drive the LEDs. 
[image]

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550
closed 2021-03-18 by UnchartedBull