UnchartedBull/OctoDash #1362: Can we remove forced repeated error popups?

CaptainSweatpants, 2021-01-31:
Can we disable these error dropdowns from permanently reaccuring? 
With the current issue with DisplayLayerProgress, I had to disable it, but now this error dropdown blocks most of the screen and any choices or selections must be made very quickly between dropdowns errors. I'd like to just have a way to acknowledge them, and have them stop after that; unless a critical error etc.

[image]

yetisnack1, 2021-02-02:
+1 and related, can we get rid of the mandatory use of displaylayerheight?

UnchartedBull, 2021-02-09:
Currently working on the next release, this will make DLP entirely optional and will replace the current layer height with current Z-Height (if DLP is disabled). This will also get rid of the error.

@CaptainSweatpants if you want to get rid of the warning all together just comment out [this line](https://github.com/UnchartedBull/OctoDash/blob/main/src/app/plugins/layer-progress.service.ts#L41). Then build & install and you should be golden. With the next update the API calls to DLP will be removed and OctoDash will rely on OctoPrint pushing new data, so (hopefully) no more errors with the next version.

closed 2021-02-09 by UnchartedBull