UnchartedBull/OctoDash #1366: Build - Can't Find Config File

Rneil-ca, 2021-02-01:
**What doesn't work?**
When trying to compile OctoDash, no matter what I try it comes up with an error: Can't Find Config File.  I must be missing something, I just can not figure out what.

**What did you already try?**
Have tried building it on Ubuntu 20 Server and Ubuntu 20 desktop and Windows 10 , I have also created deb package that I  installed on my Raspberry Pi and get the same error.  Have tried different versions of NPM as well.  I have manual placed a config.json in ~/.config/octodash/  and that does not help.

I have spent hours trying to get this work and know I must be missing something.

Thanks,
Ryan

UnchartedBull, 2021-02-01:
Please downgrade `electron-store` in the package.json to 6.0.0, 7.0.0 for some reason doesn't work. Haven't figured out yet why.

Rneil-ca, 2021-02-01:
Thanks, that did it, knew I was missing something. 

Really like what you have created, but am thinking of making some tweaks to it.  One is the M18E command for the filament change like to make it an option in the config (actually Klipper has a different command to do this, but then need to re-enable it after).  Another thing I am going to look at doing is changing the control menu to just be the action buttons, and the control panel would be included as one of those buttons.  I do not use the control buttons very often, but I use the action buttons all the time.  Plus for some reason on my touch screen, I am unable to scroll through the action items.

Have created a fork and will update, that if I have the time and actually get this all working.

Thanks again!
Ryan

UnchartedBull, 2021-02-04:
Thanks for the PR! Feel free to create more :)

closed 2021-02-04 by UnchartedBull