UnchartedBull/OctoDash #1374: Delay start plugins Encloser and DisplayLayerProgress

nettunoroberto, 2021-02-03:
Good evening, I would like to know if it is possible to delay the start of the enclosure and display layer progress plugins on boot until octodash has changed the status to "operational". I have octodash with Klipper firmware and often I have communication errors of the two plugins until octodash recognizes the connection of octoprint with klipper. 

Thank you all

UnchartedBull, 2021-02-09:
Not with the current version. The next version will rely on the OctoPrint socket, which pushes data instead of pulling it, which should get rid of your error (at least for DLP). I'll make sure to include the delay for the enclosure plugin as well.

nettunoroberto, 2021-02-14:
Tnx bro.

UnchartedBull, 2021-02-17:
Keeping this open, just so I don't forget it :) Will be closed with the correct PR.

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550
closed 2021-03-18 by UnchartedBull