UnchartedBull/OctoDash #1376: Enable Emoji Support or Use Font w/ Emoji Support (Filament names)

scottmudge, 2021-02-03:
**Is your feature request related to a problem? Please describe.**
It's fairly common to use Emoji's in Filament names to have a better indication of the color of the filament (e.g., "Hatchbox - 🟢 Green PETG"). However, the typeface/font used in OctoDash does not support emojis. It currently just shows a rectangle character.

So the Filament list in these cases looks like "Hatchbox - □ Green PETG", which isn't ideal.

**Describe the solution you'd like**
Use a typeface which supports Emoji's, or strip out emoji's or extended Unicode characters from the filament names.

UnchartedBull, 2021-02-09:
Interesting that that doesn't work. The API is correctly returning the emoji and I can put in emojis in the source code and they get shown correctly. I'll check (and hopefully fix) where they get lost. If you're interested there is also another way to let OctoDash show the color of the filament along the name: if you put the hex-color code (i.e. #FFFFFF for white) in brackets at the very end of the filename OctoDash will color the filament type with that color.

So: `spool #3 (#e67e22)` becomes:
[image]

UnchartedBull, 2021-02-17:
Are you using a custom theme? I just tested it on the dev machine and it's displaying the emoji without any issues:

[image]

scottmudge, 2021-02-18:
Using stock, with whatever comes from OctoPi. 

[image]

Pardon the blue color, just a problem w/ white balance on my camera, but you can see the emojis before the spool title are just circles. All the other colors are either filled or empty circles, but no colors/emojis.

UnchartedBull, 2021-03-02:
Mhh that is really weird. I'm going to have another look at this to check what exactly goes wrong.

cryptoAlgorithm, 2022-07-02:
@UnchartedBull Just a random guess, but could this be caused by chrome falling back to an emoji font on your dev machine, which isn't present on the pi due to the limited number of preinstalled fonts?