UnchartedBull/OctoDash #1383: System Error; getaddrinfo EAI-AGAIN octpi-local

SgtKiLLx, 2021-02-05:
i just got dash working, everything was fine. i shutdown the pi to rearrange some cords, turned it on, forgot to plug the ethernet in. shutdown, plugged the ethernet in, and now im getting this error.

running a pi4b, octodash 2.1.2

closed 2021-02-07 by SgtKiLLx

Notyly, 2021-03-16:
I have the same issue, what is the solution?

P.S. The only thing i changed is updating the firmware of the printer. Octoprint itself works fine... also the connection to the printer....

UnchartedBull, 2021-03-18:
Most likely no internet connection / your Pi didn't get a DHCPOFFER. Might be worth checking the ip from your Pi if it starts with `169.x.x.x`. You either need to manually request another IP or wait for I think 12 hours.