UnchartedBull/OctoDash #1388: Installing failed

kekko7072, 2021-02-06:
**What doesn't work?**
I tried to install using all the guides I found official and unofficial but it always shows this error on starting.

xauth:  file /root/.Xauthority does not exist

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Current Operating System: Linux octopi 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l
Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Build Date: 15 January 2021  02:03:57PM
xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support) 
Current version of pixman: 0.36.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sat Feb  6 18:35:33 2021
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.11+

or without desktop downloaded it shows screen not found.

 I dont this is linked to octodash but maybe to missing diriver or Server installation. I tried installing also desktop but nothing works, always this error and black Screen. The lcd is correctly installed in fact Raspbian desktop correctly works also using touchscreen

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.

**General Information:**
 - Raspberry Pi 3 model B, LCD https://www.waveshare.com/wiki/5inch_HDMI_LCD 
 - OctoPi and desktop support
OctoPrint version : 1.5.2
OctoPi version    : 0.18.0

kekko7072, 2021-02-06:
ERROR LOG

[    52.577] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    52.578] Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
[    52.578] Current Operating System: Linux octopi 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l
[    52.579] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
[    52.588] Build Date: 15 January 2021  02:03:57PM
[    52.588] xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support) 
[    52.588] Current version of pixman: 0.36.0
[    52.589] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    52.589] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    52.597] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Feb  6 18:48:21 2021
[    52.627] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    52.660] (==) No Layout section.  Using the first Screen section.
[    52.660] (==) No screen section available. Using defaults.
[    52.660] (**) |-->Screen "Default Screen Section" (0)
[    52.661] (**) |   |-->Monitor "<default monitor>"
[    52.684] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    52.684] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    52.685] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    52.685] (==) Automatically adding devices
[    52.685] (==) Automatically enabling devices
[    52.685] (==) Automatically adding GPU devices
[    52.686] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    52.715] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    52.715] 	Entry deleted from font path.
[    52.715] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    52.715] 	Entry deleted from font path.
[    52.716] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    52.716] 	Entry deleted from font path.
[    52.716] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    52.716] 	Entry deleted from font path.
[    52.716] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    52.717] 	Entry deleted from font path.
[    52.717] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    52.717] 	Entry deleted from font path.
[    52.717] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    52.717] (==) ModulePath set to "/usr/lib/xorg/modules"
[    52.717] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    52.718] (II) Loader magic: 0x1fdf80
[    52.718] (II) Module ABI versions:
[    52.718] 	X.Org ANSI C Emulation: 0.4
[    52.718] 	X.Org Video Driver: 24.0
[    52.719] 	X.Org XInput driver : 24.1
[    52.719] 	X.Org Server Extension : 10.0
[    52.765] (++) using VT number 7

[    52.765] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    52.767] (II) no primary bus or device found
[    52.767] (II) LoadModule: "glx"
[    52.787] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    52.922] (II) Module glx: vendor="X.Org Foundation"
[    52.922] 	compiled for 1.20.4, module version = 1.0.0
[    52.922] 	ABI class: X.Org Server Extension, version 10.0
[    52.923] (II) LoadModule: "fbturbo"
[    52.952] (II) Loading /usr/lib/xorg/modules/drivers/fbturbo_drv.so
[    52.962] (II) Module fbturbo: vendor="X.Org Foundation"
[    52.963] 	compiled for 1.20.3, module version = 0.5.1
[    52.963] 	Module class: X.Org Video Driver
[    52.963] 	ABI class: X.Org Video Driver, version 24.0
[... 173 lines cut ...]
[    63.927] (**) Option "config_info" "udev:/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1A2C:2124.0003/input/input3/event3"
[    63.927] (II) XINPUT: Adding extended input device "SEM USB Keyboard Consumer Control" (type: KEYBOARD, id 9)
[    63.927] (**) Option "xkb_rules" "evdev"
[    63.928] (**) Option "xkb_model" "pc105"
[    63.928] (**) Option "xkb_layout" "gb"
[    63.941] (II) evdev: SEM USB Keyboard Consumer Control: initialized for relative axes.
[    63.941] (WW) evdev: SEM USB Keyboard Consumer Control: ignoring absolute axes.
[    63.944] (**) SEM USB Keyboard Consumer Control: (accel) keeping acceleration scheme 1
[    63.945] (**) SEM USB Keyboard Consumer Control: (accel) acceleration profile 0
[    63.945] (**) SEM USB Keyboard Consumer Control: (accel) acceleration factor: 2.000
[    63.945] (**) SEM USB Keyboard Consumer Control: (accel) acceleration threshold: 4
[    64.046] (II) config/udev: Adding input device SEM USB Keyboard System Control (/dev/input/event4)
[    64.053] (**) SEM USB Keyboard System Control: Applying InputClass "evdev keyboard catchall"
[    64.055] (**) SEM USB Keyboard System Control: Applying InputClass "evdev keyboard catchall"
[    64.056] (II) Using input driver 'evdev' for 'SEM USB Keyboard System Control'
[... cut ...]
[    64.057] (--) evdev: SEM USB Keyboard System Control: Found keys
[    64.057] (II) evdev: SEM USB Keyboard System Control: Configuring as keyboard
[    64.058] (**) Option "config_info" "udev:/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/1-1.4:1.1/0003:1A2C:2124.0003/input/input4/event4"
[    64.058] (II) XINPUT: Adding extended input device "SEM USB Keyboard System Control" (type: KEYBOARD, id 10)
[    64.058] (**) Option "xkb_rules" "evdev"
[    64.058] (**) Option "xkb_model" "pc105"
[    64.059] (**) Option "xkb_layout" "gb"

UnchartedBull, 2021-02-09:
Have you tried the fix mentioned in this post: https://superuser.com/questions/806637/xauth-not-creating-xauthority-file? Also if Raspbian Desktop is working fine you can also start OctoDash from there via autostart.

kekko7072, 2021-02-10:
Thank you so much 
now I'll try that solution. But for the moment can you tell me how to start octodash using auto start ? Because I tried using the standard command ./octodash but it says permission denied. And if I try using sudo it says the command doesn't exist.

UnchartedBull, 2021-02-12:
just have a look at the installer script here: https://github.com/UnchartedBull/OctoDash/blob/main/scripts/install.sh#L749. There does seem to be an issue when starting via `octodash` due to a segfault, which is a known thing with electron apps.

Snip3rita, 2021-02-15:
I have problems with the same screen. I tried hundreds of tutorial now.. :(

kekko7072, 2021-03-02:
@UnchartedBull can you suggest some screen because this isn't working with OctoDash.

UnchartedBull, 2021-03-02:
https://github.com/UnchartedBull/OctoDash/wiki/Supported-Devices#recommended-screens

Snip3rita, 2021-03-02:
> @UnchartedBull can you suggest some screen because this isn't working with OctoDash.

In the end worked fine for me..I'm using a 5" Waveshare hdmi display :)

kekko7072, 2021-03-02:
> > @UnchartedBull can you suggest some screen because this isn't working with OctoDash.
> 
> 
> 
> In the end worked fine for me..I'm using a 5" Waveshare hdmi display :)

Can you explain if you did something differente from the installation guide?

closed 2021-03-21 by stale[bot]

kekko7072, 2021-03-29:
I still have this problem

kekko7072, 2021-03-29:
[image] ANY IDEA?