UnchartedBull/OctoDash #1389: Nox Theme V0.2 need help

Jack77777777, 2021-02-07:
Hi,

I copied custom-styles.css from Nox theme folder to /.config/octodash/ and overwrote the existing empty custom-styles.css. I am using a 7" 1024x600 screen. 

I have a issue and one question. 

Issue: 
The input on the loaded file screen is incorrectly arranged
[image]

Question:
I see no difference at all changing thumbnail from 220x124 to 330x186. Where should i see a difference for a file with 330x186 and the same file with 220x124?

Perhaps @noxhirsch @thegarbz @UnchartedBull or someone else can help me. 

Thanks in advance

noxhirsch, 2021-02-07:
I'll take a look! (maybe today - or at least in the next few days)

Thumbnail size should make the biggest difference in the file details view: https://github.com/UnchartedBull/OctoDash/blob/main/themes/NOX/screenshots/screenshot_file.png
But if it looks good enough for you, you can leave it at 220x124px - it doesn't *need* to be changed.

noxhirsch, 2021-02-07:
I've created a pull request for the fix :) 
Here's the fixed CSS: https://github.com/noxhirsch/OctoDash/blob/main/themes/NOX/custom-styles.css

Jack77777777, 2021-02-07:
Great! Will test later Today and report.

Jack77777777, 2021-02-07:
v0.2.1 fixed it!
And i have to correct me. 330x186 shows bigger and sharper thumnails in detail view and loaded fiel view.

Thanks for the quick fix.

closed 2021-02-09 by UnchartedBull

Jack77777777, 2021-03-28:
@noxhirsch with Beta 3.0.0 there is again something wrong with the alignment
[image]

noxhirsch, 2021-03-28:
Thank you for your notice! I'll take a look as soon as 3.0.0 gets officially released, since I don't have beta installed and it's not a severe problem - just ugly - and everything is still readable ;)

Jack77777777, 2021-03-29:
Cool. I think official relese will be this or next week. For me NOX is the best theme ATM. thanks for your great work!

thegarbz, 2021-03-31:
Just realised @noxhirsch you now have a horizontal progressbar in your 0.21 theme. Kinda makes mine redundant now 🙂