UnchartedBull/OctoDash #1391: Black screen with white cursor at top left

deltwalrus, 2021-02-07:
**What doesn't work?**
After installing Octodash via the automatic install, and several hours of troubleshooting i am no further forward.
The pi boots fine and i can see the output on the screen then i goes to a black screen with a non blinking cursor in the top left hand corner.
I'm able to connect to the pi via ssh and it responds to commands.

**What did you already try?**
Attempted steps in https://github.com/UnchartedBull/OctoDash/issues/470
Copied sample files from Waveshare git repo to correct locations:
- LCD-show/etc/X11/xorg.conf.d/99-calibration.conf-35b-180 (as my screen needed flipped) -> /etc/X11/xorg.conf.d/99-calibration.conf
- LCD-show/usr/share/X11/xorg.conf.d/99-fbturbo.conf -> /usr/share/X11/xorg.conf.d/99-fbturbo.conf
- LCD-show/usr/share/X11/xorg.conf.d/45-evdev.conf -> /usr/share/X11/xorg.conf.d/45-evdev.conf

**General Information:**
 - Hardware: Raspberry Pi 3B, [Waveshare 3.5" LCD](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B))
 - OS Info: OctoPi 0.15.1
 - OctoDash Version: latest from git
 - OctoPrint Version: 1.5.3

deltwalrus, 2021-02-08:
Also ran through all steps listed at https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#cannot-open-display-xinit-failing-

deltwalrus, 2021-02-08:
No output via HDMI, monitor says "input timing is not supported" (but I am getting console through the LCD).

deltwalrus, 2021-02-09:
Is it possible that certain Octoprint plugins might be causing this to fail to launch?  Is there a log file that I can look to for some clues?  The X server log has no errors.

deltwalrus, 2021-02-09:
Well, without any other ideas of what to try, I am now wiping my Octoprint installation and starting over with no plugins installed whatsoever, save for what Octodash wants to bring along.  We'll find out that way if there were plugins that conflict with this software.

deltwalrus, 2021-02-09:
1. Wiped SD card
2. Re-applied latest OctoPi image
3. Changed WPA2 to be able to login remotely
4. git clone https://github.com/waveshare/LCD-show.git
5. cd LCD-show
6. chmod +x LCD*
7. ./LCD35B-show (system reboots and I get a console display)
8. ./LCD35B-show 180 (system reboots and I get a console display inverted 180 degrees to the orientation I want)
9. Installed Octodash as per instructions, selecting Display Layer Progress (mandatory), Preheat Button, Print Time Genius, PrusaSlicer Thumbnails plugins
10. Allowed Octodash to start at boot
11. Setup update script
12. Rebooted pi
Stuck at console login, so...
13. sudo raspi-config, Option "3 Boot Options", Option "B1 Desktop / CLI", Option "B2 Console Autologin"
14. Finish and reboot
X errors about no screens found abound, so...
15. copied above-mentioned X config files from git repo to local directories
16. add config info from https://www.raspberrypi.org/forums/viewtopic.php?t=162323

System reboots and comes up still with blank screen and non-flashing cursor in top left corner.  I do not know what I am doing wrong at this point.  Help, please?

deltwalrus, 2021-02-09:
Okay, I saw it *briefly* flash the "Hey there!" intro screen for a fraction of a second, so maybe this is a resolution issue.  I will continue to play with resolutions to see what I can figure out.  Any guidance?

UnchartedBull, 2021-02-09:
Great for trying so many things and sorry that none have worked out yet!

First off the OctoPrint plugins won't prevent OctoDash from starting, so you should be fine on that front.

The screen you're using is connected via the GPIO pins right? These are known to be tricky to setup, but they should work fine once you get up and running (do make a clean backup though :D)

I retired my 3.5" GPIO screen a while back so I won't be of much help here but here are a few things I would try:

- Do you see the white ratpoison box in the upper right corner for a few seconds directly after booting finished? You're also mentioning that there is only a white cursor at the very top, so the normal console output that is scrolling through while booting is replaced by that?
- Here are some other issues with similar problems, maybe one of those work for you: #500, #574, #1039 plus probably a few more buried somewhere deep in the closed issues
- As a last resort I would install the fully fledged Raspbian Desktop. If the screen works there you can install OctoPrint and then manually install OctoDash (skipping all the XOrg parts and don't install ratpoison). You can then start OctoDash via the autostart menu in Raspbian.

Let me know if any of these work for you :)

deltwalrus, 2021-02-09:
I gave up on this screen, I am returning it and ordering a new one with HDMI output as opposed to going through GPIO.  Hopefully that will help.

> Do you see the white ratpoison box in the upper right corner for a few seconds directly after booting finished? You're also mentioning that there is only a white cursor at the very top, so the normal console output that is scrolling through while booting is replaced by that?

The system boots with normal console output, then a blank screen with a white cursor appears.  Nothing else appears.

>Here are some other issues with similar problems, maybe one of those work for you: #500, #574, #1039 plus probably a few more buried somewhere deep in the closed issues

I gave those a read as well, nothing in there seemed to help.

>As a last resort I would install the fully fledged Raspbian Desktop. If the screen works there you can install OctoPrint and then manually install OctoDash (skipping all the XOrg parts and don't install ratpoison). You can then start OctoDash via the autostart menu in Raspbian.

I did run the desktop installation, no change in behavior.  I really think this screen is just hot garbage, so I will give it another try in a few days when the new screen arrives.

A suggestion, if I may:  put a Google spreadsheet out there where folks can denote what specific screens they used, along with what additional steps they had to take to get it working with this project.  I realize that has nothing to do with Octodash itself, but it would go a long way toward making setup less painful.

UnchartedBull, 2021-02-09:
I think you'll be happier with an HDMI screen, plus you get these juicy 30fps animations instead of the 1-2 Hz refresh rate on GPIO monitors. The google doc for screens is actually a great idea, there already is a list of screens that are known to be working good available in the [wiki](https://github.com/UnchartedBull/OctoDash/wiki/Supported-Devices#recommended-screens) but adding the extra steps necessary might be a nice addition here.

If you haven't packed up the screen to return yet there might be one more thing worth trying: https://github.com/UnchartedBull/OctoDash/wiki/User-Guides#uctronics-lcd35-spi-install

mzilly02, 2021-02-18:
> Great for trying so many things and sorry that none have worked out yet!
>
> First off the OctoPrint plugins won't prevent OctoDash from starting, so you should be fine on that front.
>
> The screen [...]

Hi
I'm having the same problem using a raspbian i386 VM in virtualbox.
I installed octoprint in /home/pi//octoprint/venv/... following this guide: https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian-or-raspberry-pi-os/2337 .
I installed Octodash following this guide: https://github.com/UnchartedBull/OctoDash/wiki/Installation .

**What doesn't work?**
Using the xinitxrc provided it went to CLI and octodash doesn't start saying "libx11-xcb.so.1 cannot open shared object file: no such file or directory".

**What did you already try?**
Modifying the xinitxrc with this:
`   #!/bin/sh

    xset s off
    xset s noblank
    xset -dpms
   
   export DISPLAY=:0 ratpoison&
    octodash
`

ratpoison shows briefly like in the screenshot ( [image] ).
Octodash shows a black screen.

**General Information:**

- Hardware: Virtualbox 6.1.6 4 cores w/4gb ram
OS Info: RaspbiOS buster i386 2021-01-11
OctoDash Version: latest from git
OctoPrint Version: latest from pip install octoprint

UnchartedBull, 2021-03-02:
Ratpoison will only show the banner for 2-3s and then return to a completely black screen. This normally means that everything is working and OctoDash should start. 

Could you change the last part of your `~/.xintirc` to this and check if that works?

```
export DISPLAY=:0&
ratpoison&
octodash
```

thegarbz, 2021-03-03:
@mzilly02 You're pretty deep in the weeds here I'm not sure what you think you're running but it's unlikely to be Raspbian i386 since there's no such distro release. Raspbian is purely an ARM based release. Likely you're running Debian with the Raspberry Pi Desktop installed. Distinction is that it ships with a different set of packages and dependencies. 

Try installing the missing library: sudo apt-get install libx11-xcb1
If that fails try force installing the i386 version: sudo apt-get install libx11-xcb1:i386

The internet is full of problems with libx11xcb and all sorts of software running on a mix of ARM, x64 and i386 distros. 

@deltwalrus sorry I didn't see this earlier. I had a Waveshare screen and I assure you I had no end of difficulties getting a working X configuration working for it.  After a cool week or two I did eventually get it working only to get frustrated by the very slow update rate on the display, and abandoned it in favour of a HDMI display. You definitely made the right call, the 3.5" waveshare display is a pain to use. I don't know why these damn waveshare displays are so popular, there's no end of problems with them on various raspberry pi setups.

deltwalrus, 2021-03-10:
I'm chalking this up to the original display being hot garbage, even though I saw anecdotal evidence of it working for others. New display worked without a hiccup.