UnchartedBull/OctoDash #1392: Filament Change: Purge before unload

Jack77777777, 2021-02-08:
Hi There,

I have a bowden and if i don't Purge (10mm for example) before unloading filament there is a blob and filament stucks in my extruder. Is there a way to let Octodash purge before unloading?

UnchartedBull, 2021-02-09:
No unfortunately there isn't. Is that supported by M600 by any chance?

Jack77777777, 2021-02-10:
@UnchartedBull Thanks for the feedback!

I can setup custom action commands for load/unload and use the planned feature from #1291
`hold on a filament for 3 seconds and it will directly set the new filament, no heating, no other commands`
Can you integrate two customs actions in the Filament Change Menu?

Procedure would be then:
1. Unload using custom action
2. Load using custom action
3. Hold new Filament 3 seconds to change in Octoprint

UnchartedBull, 2021-02-12:
The custom actions are fully user configurable, so you can add there whatever you like https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#configuring-custom-actions

You can already test the new filament fast-track with the latest beta available in the discord.

closed 2021-02-12 by UnchartedBull

Jack77777777, 2021-02-12:
I have already configured the custom actions for unload/load and they are working from control panel of octodash. I was asking if you can implement custom actions in the filament screen on octodash that all is together and there is no need to jump to control and then to filament screen