UnchartedBull/OctoDash #1398: XInit failures with Official RPI touchscreen

joel-kelly, 2021-02-10:
**What doesn't work?**
I'm getting xinit failures and none of the steps listed in the xinit failure troubleshooting guide have worked for me. I get to a black screen with the "welcome to ratpoison" message before it boots me back to the CLI-- see log pasted below.

When I try to run "ratpoison" I get "Can't Open Display".

I'm using the official RPI touchscreen, and have verified that it's working in the Raspbian desktop GUI. Are there some other drivers that I'm missing for this?

**What did you already try?**

- All of the steps listed in the [Xinit troubleshooting guide.](https://github.com/UnchartedBull/OctoDash/issues/574#issuecomment-621324820)
- Tried wiping the SD card and reinstalling following the steps in sequence listed by [this closed issue](https://github.com/UnchartedBull/OctoDash/issues/574#issuecomment-621324820)
- export DISPLAY=:0 before running xinit / sudo xinit

**General Information:**
Hardware: Raspberry Pi B+, official Raspberry Foundation 7" Touchscreen
OS Info: OctoPi 0.17.0
Octodash version: latest from git (installed 2021-02-9)
Octoprint version: 1.3.12

CLI output when xinit boots me out:
`X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Current Operating System: Linux octopi 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l
Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 15 January 2021  02:03:57PM
xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Wed Feb 10 01:03:05 2021
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.11+
Illegal instruction
xinit: connection to X server lost

waiting for X server to shut down XIO:  fatal IO error 4 (Interrupted system call) on X server ":0"
      after 159 requests (159 known processed) with 0 events remaining.
(II) Server terminated successfully (0). Closing log file.`

UnchartedBull, 2021-02-12:
Do you mean the OG Raspberry Pi 1 with Raspberry Pi B+? If so that Pi isn't supported by OctoDash (there is no armv6 support in Electron) https://github.com/UnchartedBull/OctoDash/wiki/Supported-Devices. That would also explain the illegal instructions, since you would be trying to execute armv7l code on the armv6 processor, which it doesn't understand.

joel-kelly, 2021-02-12:
Oh butts you're totally right 🤦. I guess I got confused by the 1/2/3 vs the A/B/C naming convention. Sorry!

For anyone else that gets this message, please check that you're using the right Pi!

closed 2021-02-12 by joel-kelly