UnchartedBull/OctoDash #1399: PSU control - catch22

SaKiEQ, 2021-02-10:
howdy, I configured PSU control and got a lovely working Power Button in the action commands pallet. 

Only thing - catch22 - the moment I turn the printer PSU off, well it goes off... with it the printers control board, with that
the USB connection to OctoPrint, with that OctoDash goes into 'sleep mode'. 

Now I am locked-out from the action commands screen to turn the printer back on ;), since we are offline.

DisasteR, 2021-02-11:
normally if you press on the screen during sleep and have enabled psu-control in octodash it will wake up your printer.

SaKiEQ, 2021-02-11:
have to try that... this idea / behavior hasn't crossed my mind. All I noticed was that with turning off the power the printer disconnected.

Still I think it would be nice if the control panel would be accessible during printing, with the manual arrows disabled. 
If somebody wants to hit heat-up or cool-down, homing or any other function during the print, then that is that. 
The same happens during printing vial Octoprint if somebody manipulates the printer from its built-in controls, they dont disable either ! 

After all the touch screen / OctoDash is my 99% way of interacting with the printer, I could as well remove the built-in controls.

UnchartedBull, 2021-02-12:
Custom Actions on sleep screen will be included in the next release (already available on beta). Custom Actions during printing will be something for the release after.

closed 2021-02-12 by UnchartedBull

SaKiEQ, 2021-05-22:
Hi there, 

first off, great job on the new side drawer fur actions  while in sleep screen. 
While I have now the PSU control working, my custom action script to control lights, 
attached to M118 GCODE commands, can not be executed when OctoDash is idling. 

One method is to turn the printer PSU on, then the light on/off. 

While the PSU operates with 'M' GCodes too, why would that work and not the M118 ?