UnchartedBull/OctoDash #1416: M600 disconnect octoprint from printer

jpokorny12, 2021-02-11:
Hello 
I have 3,5 tft display with octodash connected to skr e3 mini.
When I send M600 to printer, octoprint disconected from printer. Terminal show this.

I try install octoprint without octodash to raspberry and when I send M600 to printer, everything works. 
When i install octodash on clean instalation octoprint, M600 disconect octoprint.
I think the M600 doesn't work with octodash, because when octodash isn't installed it works.

In Marlin i have uncomend host_action_commands.

```
Changing monitoring state from "Pausing" to "Paused"
Recv: //action:paused
Printer signalled that it paused, switching state...
Recv: T:200.07 /200.00 B:48.44 /50.00 @:51 B@:0
Recv: T:200.10 /200.00 B:49.46 /50.00 @:51 B@:127
Recv: T:200.00 /200.00 B:49.87 /50.00 @:54 B@:127
Recv: T:200.03 /200.00 B:51.82 /50.00 @:53 B@:0
Recv: T:200.00 /200.00 B:51.82 /50.00 @:53 B@:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Paused" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor
```

UnchartedBull, 2021-02-12:
Have you tried adding M600 to long running commands like described in the error log? https://community.octoprint.org/t/octoprint-keeps-running-into-communication-errors-and-timeouts/227

jpokorny12, 2021-02-12:
Yes, I have M600 in long running commands.

[image]

jpokorny12, 2021-02-18:
[serial.log](https://github.com/UnchartedBull/OctoDash/files/6005637/serial.log)
[octoprint.log](https://github.com/UnchartedBull/OctoDash/files/6005638/octoprint.log)

closed 2021-02-22 by jpokorny12

jpokorny12, 2021-02-23:
It is marlin isuue ;), sorry
https://github.com/MarlinFirmware/Marlin/issues/21143
Thank you 👍