UnchartedBull/OctoDash #1418: (Offline) Serial Exception Error

911leo, 2021-02-11:
**What doesn't work?**
When using OctoDash, I receive an Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:readiness:3831) during printing. I have seen a lot of people having this issue regarding OctoPrint, but not specifically with OctoDash.

I can disconnect the screen I use for OctoDash, leaving only the USB serial connection to the printer and I don't receive any issues when printing from OctoPrint.

If I print from OctoPrint while the screen is connected, I will still receive the error.

**What did you already try?**
Taped the 5v pins on the USB cables to make sure it wasn't pulling power
Different USB cable
Used different USB Port (thought this worked for a moment but same error at 77% complete)

**General Information:**
 - Hardware [Raspberry Pi 4b 8GB - Ender 3 v2]
 - OS Info [Raspian Buster Lite 0.18.0 - OctoPi: Yes (file: octopi-buster-armhf-lite-0.18.0)]
 - OctoDash Version [v2.1.2]
 - OctoPrint Version [v1.5.3]

**Additional context**
Added my octoprint.log file, I just realized that I didn't have serial reporting turned on. But I will try another print with it on.
[octoprint.log](https://github.com/UnchartedBull/OctoDash/files/5968856/octoprint.log)

911leo, 2021-02-12:
I don't think this is an OctoDash issue anymore.

I just installed OctoDash last night, but used a different SD card than I was originally using, so everything had to be installed again.
I began having the issues every time I tried to use OctoDash, so as I said above I just unhooked the screen and went back to running through OctoPrint. Everything last night printed fine so I assumed that it was an issue with OctoDash.
After numerous failed attempts to print the same piece today with OctoDash I decided to reboot without anything hooked up except for the Pi to the printer. I began the print again from OctoPrint and it failed for probably the 9th time today.

Now, I am seeing it fail while I am only using OctoPrint without the screen attached. Prints will stop randomly for a second or two which leaves a small spot then continue on until the serial connection problem.

Finally gave up and printed the file from the SD card and it printed perfect.
Time to start over from scratch and try again.

closed 2021-02-12 by 911leo

UnchartedBull, 2021-02-12:
Could be that your USB port is running out of bandwidth. There are options in marlin to up the baudrate so you have more headroom plus there is a new plugin available which compresses GCode before sending it over to the printer. Might be worth a try if you're running Marlin: https://plugins.octoprint.org/plugins/meatpack/ (I haven't tested it, so do it your own risk)

joann235, 2021-10-01:
I get this error whenever I attempt to print a file:

State: Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831)

Raspberry Pi 4 with raspberry pi camera running Version 1.5.2

I have verified USB cable connection.  Prior tot trying to print file it reports connected and operational. I have loaded and reloaded Octopi software at least 4 times always getting same error.  Any help would be appreciated.

UnchartedBull, 2021-10-01:
Does this happen when OctoDash isn't running? This error seems like something is wrong with your setup and not necessarily with OctoDash. Please try a print without OctoDash and report whether the issue is still occurring.

joann235, 2021-10-01:
I apologize but I dont know how to do what you suggest..  What is octodash?  I have the GUI that I expected with graphs nd file selection.  The status reports operational and connected.  I am not aware of any settings that I had to select to get going.

I really appreciate how quickly you responded.  Thank you.

jneilliii, 2021-10-01:
> What is octodash?

OctoDash is the project/software that you are replying your issue to. If you are experiencing this with OctoPrint you should connect on the community forum or Discord server.