UnchartedBull/OctoDash #1421: Only getting to the Pi Login Screen

tre4b, 2021-02-12:
**What doesn't work?**
No sign of OctoDash at all.  The pluggins requested during the install are installed though so I think it should be.  The Pi Boots fine and ends up sitting at the standard login prompt on the 3.5" screen.

**What did you already try?**
I checked the ./xinit and it octodash is there.
Did a reboot then 
    sudo ps -A | grep "octo"

this returned octoprint but no octodash so looks like it is not running.

looked for a log in /var/log

I am running headless except for the 3.5" screen.   When I have ssh'd in I ran

sudo octodash &    

I was hoping it would know about the display as I did not know how to tell it to point at the 3.5 display what I got was.

[6112:0212/210205.353735:FATAL:electron_main_delegate.cc(254)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.

not at all sure how that would happen.

**General Information:**
 - Running on a Pi4 with 4GB ram,  thus need to use a 3.5 with pins because of the mini hdmi stuff :-(
 - Whatever is your most up to date, downloaded using 
 - bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
 - OctoPrint Version 1.5.3
 - From the most current OctoPi image 0.18

**Additional context**
I've tried to set this up a couple of ways to get the 3.5 working.  This time I set it up using the instructions for OctoScreen up to the point where it talks about installing OctoScreen.   At that point I installed OctoDash on this sd card.  On a duplicate I installed OctoScreen and verified it worked.  (what can I say I wanted to see which I prefered).  I reasoned because the display worked for OctoScreen it ought to work for OctoDash too.

Demannu, 2021-02-13:
So, the error you're getting is because you're running as sudo. Octodash can operate at a user level. Try running without the sudo.

I got the same error, but after dropping the sudo I'm getting a segmentation fault when it tries to launch. I'm filling an issue for that now

tre4b, 2021-02-13:
@Demannu Yeah sorry I did do it without Sudo too.  That resulted in no process running,  and yes it does come up with a Segmentation fault same as you without sudo.   Damned if you do, damned if you don't ;-)

Icesythe7, 2021-02-13:
I had these errors I just had to change setting to auto login

tre4b, 2021-02-14:
I'll give that a go.  Last time I did this when first trying to setup OctoScreen it all went to pot ;-)  Hopefully should work,  makes sense if OctoDash runs in user context.  Printing right now though so will have to wait a bit ;-)

tre4b, 2021-02-14:
Yep that did it.  Many thanks  Icesythe7

closed 2021-02-14 by tre4b