UnchartedBull/OctoDash #1425: Failure response for Displaylayerprogress

Icesythe7, 2021-02-13:
**What doesn't work?**
I keep getting the tittle error every 5 mins or so yet it displays the layers and seems to work flawlessly, however this keeps popping up and being annoying.

**What did you already try?**
Googled the issue but can't find an istance where everything seemingly works as intended but get random (fake?) errors.

**General Information:**
 - Hardware rpi 4 8gb
 - OS Info octoprint 32bit latest stable
 - OctoDash Version 2.1.2
 - OctoPrint Version 1.5.3

**Additional context**

UnchartedBull, 2021-02-17:
Duplicate of #1304.

UnchartedBull, 2021-02-17:
Whoops, I was a little fast there. If the errors are only occurring occasionally you might want to adjust your poll time in the config.json (maybe something like 3000). Nevertheless this will be fixed with v3.0.0 as there will be no more regular API calls which can produce errors.

closed 2021-02-17 by UnchartedBull

Icesythe7, 2021-02-18:
> Whoops, I was a little fast there. If the errors are only occurring occasionally you might want to adjust your poll time in the config.json (maybe something like 3000). Nevertheless this will be fixed with v3.0.0 as there will be no more regular API calls which can produce errors.

3000 didn't change much, will just wait on the v3 release thanks for the response regardless!