UnchartedBull/OctoDash #1435: Power Off / reboot Pi button.

midmadn, 2021-02-17:
I love and use Octodash a lot. The biggest wish I have is to have a red power button on the main screen that when you press it you can choose to power down the Pi or reboot the Pi. Especially the power down. I constantly switch the power off which I've read is bad to do - just like in windows. Any hope for this to be added?

ubermick, 2021-02-17:
Adding my vote to this - would be a great addition. I only power my printer on when I need it, so would prefer to not have the Pi and camera running all the time. (I mean, I know I can shut it down via SSH, but this would be SOOOO much easier!)

ubermick, 2021-02-17:
...OR I could search and see it's already possible!

https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions

(Be nice to have it included and enabled by default, though!)

UnchartedBull, 2021-02-17:
The skull in the bottom right corner does exactly that. Probably a good idea to change it to a power icon by default though.

commit adc3f868 (2021-02-17): replace default config kill with shutdown

fixes #1435
closed 2021-02-17 by UnchartedBull