UnchartedBull/OctoDash #1436: Allow selecting filament WITHOUT sending ANY G-Code.

scottmudge, 2021-02-17:
**Is your feature request related to a problem? Please describe.**
Even with M600 mode set, this is the 3rd time now that using the filament change mode has just rammed the extruder of my printer into the endstop. For some inexplicable reason, after pre-heating the extruder is moved in *relative* units about +50mm in the X-axis. Not *absolute* units, which would make way more sense if the goal is to get the extruder somewhere accessible in the middle. Yeah fine, relative units work if the printer isn't homed. But it always is in my case, and the genius of using relative units is that if the extruder is already close to the end of the X-axis print volume, it doesn't stop. It just keeps ramming itself into the end, causing stepper noise and bearing damage. So that's really great.

It was suggested to use M600 mode before, which I am, but it's still prone to causing all sorts of problems.

All I want to do is change the active filament being tracked. That it. I don't want it to do anything to my printer. I don't trust the plugin or OctoPrint to do it correctly, because it frequently doesn't. 

How can I disable ALL outgoing g-code for filament changes? I don't want it to send *anything*. I just want it to change the active filament being tracked in OctoPrint.

UnchartedBull, 2021-02-17:
As already mentioned in your last issue, the next release will include the fast track to change filament without any GCode being send (it is already available in the beta).

I don't really know why your printer is moving the X-Axis during filament change, since OctoDash is only sending commands to the extruder.

closed 2021-02-17 by UnchartedBull

scottmudge, 2021-02-18:
Is there a nightly build or branch with this change, or is it still pending?

And I don't know why either, looking through all of the OctoPrint configurations, I don't see any script or g-code associated with tool change or m600 commands.

And the firmware sets an absolute x-position for M600 commands without an explicit x-shift:

[image]

For now I'm going to modify the firmware directly and prevent it from moving in any axis except E and Z while not printing, and if M600 is sent during the middle of a print, it will move to a safe absolute X-position.

Thanks for the help.

scottmudge, 2021-02-20:
Alright I modified the firmware to behave correctly w/ M600 commands, but after installing Midnight's nightly build (Snapshot 02-06-2021), the M600 command is no longer being sent by OctoDash, even with it enabled in the config. So unfortunately I got what I desired in the end, but I was generally enjoying the OctoDash M600 implementation combined with my firmware changes.

I also noticed the configuration validation wizard had issues validating the presence of "disableExtruderGCode" and "screenSleepCommand", though I confirmed they existed in the config.json file.

I restarted the machine, and it re-initialized a blank config, which was a little bit annoying, but it's a short config to set up.

Anyway, here's my config params. Does setting the "feedSpeed" or "feedLength" to 0 disable the M600 being sent? The prompt continually noted "M600 command sent", but the printer did nothing until I manually sent "M600" in OctoPrint's terminal.

OctoDash seemed to be communicating with the printer otherwise, as the temperature command worked just fine.

[image]

UnchartedBull, 2021-03-02:
No M600 should be send either way. I did a complete rewrite of the backend logic for that part after the last release so something probably got broken in the process. Going to check that now.

UnchartedBull, 2021-03-05:
[image]

There's the problem. Forgot to put that line back in. Should be fixed with #1438.