UnchartedBull/OctoDash #1445: rgb status

jeinteriordesign, 2021-02-19:
**What doesn't work?**
After octodash install the rgb status does not work

**What did you already try?**
reinstalling rgb status in octoprint wire check

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2021-03-02:
Please link the other project.

thegarbz, 2021-03-03:
Can you provide some information? What's not working? Doesn't work in general? Error messages? Did you have a display on your raspberry pi before and is it a GPIO display or a HDMI? 

I've been using RGB Status since the beginning and never come across any conflict with Octodash.

cp2004, 2021-03-06:
I maintain [WS281x LED Status](https://plugins.octoprint.org/plugins/ws281x_led_status) - which does everything RGB Status does - usually the issues I get through supporting users are related to the touchscreen hardware conflicting or taking over the SPI connection to the LEDs.

I doubt it is an OctoDash issue, far more likely it will be OS configuration or the hardware of the screen interfering.

UnchartedBull, 2021-03-18:
Thanks @cp2004. MIght need to check out your project for my next printer, the breathing or pulsing while printing should look really nice on a small NeoPixel ring :)

Closing this here, if the issue still persists please reopen.

closed 2021-03-18 by UnchartedBull