UnchartedBull/OctoDash #1446: Octodash stuck on connection screen and cant back out, just says "connection timed out"

Jasontwd, 2021-02-19:
Hi,
steps
1. I installed latest version octoprint 0.18.0 and flashed etc
2. Setup wifi config
2. SSH into Rpi with touchscreen connected
3. Loaded script for octodash and plugins.
4. rebooted
5. loaded Octprint GUI and checked connection to printer and moved axis
6. Followed setup on touchscreen to setup connection etc.

all worked well.

Then machine had to reboot after shutdown for some reason, under power, but fixed that.

Now i get just the connection screen and time out error.

I can still use the server via GUI octoprint address and all connects ok. Just see the below image on screen.

**Any fixes for this, The discord server, A guy mentioned that:**
**_the 502 bad gateway errors is probably because Octodash loads faster  then Octoprint and it was not communicating on the initial phase._**
cheers
jason

[image]
[image]

rjbrittain11, 2021-02-19:
I'm having the same exact same issues.  All worked well initially, then at some point after a restart of the printer, i now get the same connection time out message.  I am connected to OctoPrint via my IP in Chrome and can do anything and everything there with no problems.  I've pressed the screen multiple times in a row once OctoPrint GUI running in my browser and it trys to connect, then give the same time out error over and over.

Jasontwd, 2021-02-20:
Hi @rjbrittain11 , I found the solution work around.
SSH into the Rpi.
Once login, you will see a command prompt screen, put your username and password. 
Then put this code: **rm .config/octodash/config.json**
It basically removes the config file of octodash.
Restart the PI: Sudo reboot
Once it has been restarted, you should see the Octodash splash screen and setup menu again
go thru the setup again.
Go into general settings in octodash screen and set the Octoprint connection to your actual IP address, eg 192.168.1.***
not "octoprint.local"
Then save.
It should now be ok once the Rpi server connects
jas

rjbrittain11, 2021-02-21:
@Jasontwd you are awesome!  That worked perfectly.  I've been trying to figure this out for ages!  Thank you.

UnchartedBull, 2021-03-02:
Please also try the fix mentioned by @Jasontwd and reopen if the error still comes up.

closed 2021-03-02 by UnchartedBull

locki-cz, 2021-07-18:
better is just edit that config file then go throught config again :)

cd ~/.config/octodash/
nano config.json

and change "url": "http://raspberrypi.local:5000/"    rapberrypi.local to your local ip 192.168.x.y