UnchartedBull/OctoDash #1447: Initial install Feed Length and Feed Speed buttons do not work.

nickm324, 2021-02-19:
**Describe the bug**
During initial install and going through the on screen setup when you get to the screen that wants to set the Feed Length and Feed speed the buttons "+" and "-" do not work at all.
All other touch screen functions work fine and you just have to skip that screen during the first time run.

**To Reproduce**
Steps to reproduce the behavior:
1. Do a new install.
2. On the setup screen try using the plus or minus buttons for feed length or feed speed.
3. no response numbers do not increment or decrement.

**Expected behavior**
expected to be able to set the values using the onscreen buttons.

**General Information:**
 - Raspberry Pi 3B
 - OS Info  Raspbian Buster / OctoPi]
 - OctoDash Version  v2.1.2
 - OctoPrint Version Version 1.5.3

UnchartedBull, 2021-03-02:
Oh that's interesting, what touchscreen are you using / or were you using a mouse? Those buttons won't work with a mouse unfortunately since they have handles for long pressing, which did have some compatibility issues with handling touch & mouse events.

closed 2021-03-18 by UnchartedBull

nickm324, 2021-03-18:
Sorry, never got a notification this had been updated with a post.
In any case I am using a touch screen no mouse. Its  a Smraza 3.5 Inch Resistive Touch Screen TFT LCD Display 480x320.
No mouse.

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550